mne-tools/mne-python #2069: FIX: GAT.plot() GAT.plot_diagonal()

kingjr, 2015-05-07:
I think there s room for improvement in the GAT plotting function regarding
* default timing of the matrix: It seems that it starts at 0 even when the time starts at 100ms. I think it may be due to a ```plt.imshow``` issue.
* default ylim of the plot_diagonal(): currently it is [0 1] by default, which is ok for accuracy and auc, but  not all scorers. I suggest we just use the min and max score instead.
* chance level is by default .5. It should instead adapt to the number of classes and/or be given as a parameter for cases where it's difficult to deduce chance level (i.e. ad hoc classifiers, continuous scoring etc).
* color scale of gat.plot() only works for 2 classes.

@cmoutard. Do you have any other thoughts on this ? 

I'll deal with this FIX once #1965 is done.

cmoutard, 2015-05-09:
For now, I have nothing in mind. Thanks !

kingjr, 2015-05-21:
All fixed in https://github.com/mne-tools/mne-python/pull/2072 but the colors for > 2 class.

closed 2015-05-21 by kingjr