mne-tools/mne-python #2071: identify channels in butterfly plot

teonlamont, 2015-05-08:
It would be great if there would be a way to identify a bad channel in the topo butterfly plot. sometimes you just don't know how bad a channel is until you plot it among the others.

agramfort, 2015-05-08:
You mean click a line in evoked plot and get channel name?

dengemann, 2015-05-08:
> It would be great if there would be a way to identify a bad channel in the topo butterfly plot

for that just don't pick it away don't plot with exclude='bads', and it will be displayed.

teonlamont, 2015-05-08:
@agramfort yes, that.

Eric89GXL, 2015-05-08:
http://matplotlib.org/1.4.0/users/event_handling.html#object-picking

I might try getting it working this morning, and also making it so that our `raw` plotting uses this instead. IIRC right now it does something with display ranges, which doesn't seem as elegant.

PR #2073 (2015-05-10): ENH: Add butterfly channel picking
Closes #2071.
Closes #2015.

Ready for testing / review / merge @teonlamont.
closed 2015-05-10 by Eric89GXL