mne-tools/mne-python #2082: Connectivity matrix for sensor-space based spatio-temporal clustering

teonlamont, 2015-05-11:
Is there currently a way to calculate a connectivity matrix of the sensors for systems that don't have neighborhood matrices in `mne.channels.data.neighbors`?

dengemann, 2015-05-11:
Nope. See Fieldtrip docs for additional info on how to do that.
Or define a layout by hand (seriously).

teonlamont, 2015-05-12:
but why do we offer sensor-space spatio-temporal clustering if we don't support it for all the systems we support? I just went over to fieldtrip and they offer three different methods:

```
FT_PREPARE_NEIGHBOURS finds the neighbours of the channels based on three
  different methods. Using the 'distance'-method, prepare_neighbours is
  based on a minimum neighbourhood distance (in cfg.neighbourdist). The
  'triangulation'-method calculates a triangulation based on a
  two-dimenstional projection of the sensor position. The 'template'-method
  loads a default template for the given data type. prepare_neighbours
  should be verified using cfg.feedback ='yes' or by calling
  ft_neighbourplot
```

maybe we should port over their triangulation method since we have the capability to project sensor data to 2d.

teonlamont, 2015-07-24:
connectivity matrices added for KIT. currently this is handled in fieldtrip and ported over

closed 2015-07-24 by teonlamont