mne-tools/mne-python #2083: problem running plot_cluster_stats_spatio_temporal.py  example

atalanti, 2015-05-11:
Hello,
I am trying run the given example "plot_cluster_stats_spatio_temporal.py" which is located into mne-python/examples/stats .
When I run it in IPython terminal I take this error: 

TypeError                                 Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    202             else:
    203                 filename = fname
--> 204             __builtin__.execfile(filename, *where)

/home/atakoulini/Documents/mne-python/examples/stats/plot_cluster_stats_spatio_temporal.py in <module>()
    162 stc_all_cluster_vis = summarize_clusters_stc(clu, tstep=tstep,
    163                                              vertices=fsave_vertices,
--> 164                                              subject='fsaverage')
    165 
    166 #    Let's actually plot the first "time point" in the SourceEstimate, which

TypeError: summarize_clusters_stc() got an unexpected keyword argument 'vertices'

When I run it in the tsch terminal I get this similar error:
...
Computing cluster p-values
Done.
Visualizing clusters.
Traceback (most recent call last):
  File "plot_cluster_stats_spatio_temporal.py", line 164, in <module>
    subject='fsaverage')
TypeError: summarize_clusters_stc() got an unexpected keyword argument 'vertices'

I checked and I have the last update. So there is 'vertices' into the function cluster_level.py/summarize_clusters_stc

What do I do wrong?
Thank you,
atalanti

agramfort, 2015-05-11:
you must be using the stable version of MNE python and use the development
website to get inspiration.

atalanti, 2015-05-11:
Actually I am using the last master version of mne-Python.
I found the example from http://martinos.org/mne/stable/auto_examples/index.html

Does this mean that I cannot run the specific example? I wanted to run it and see how it works in order to do something similar for my data.  I have tested some other examples too from this link (not in the Statistics Examples category) and I did not have any problem.
Thank you in advance,
atalanti

dengemann, 2015-05-11:
checkout  http://martinos.org/mne/dev/auto_examples/index.html
<http://martinos.org/mne/stable/auto_examples/index.html>
you're trying to run outdated code.

atalanti, 2015-05-11:
You were right. I was trying to run outdated code.
Thank you

closed 2015-05-16 by Eric89GXL