mne-tools/mne-python #2084: random_state with infomax

teonlamont, 2015-05-11:
Currently when I run ica with infomax with a random state set, my results are different.

```python
ica = mne.preprocessing.ICA(.9, random_state=42, method='infomax')
ica.fit(raw, decim=4)
```
two separate runs of this code (pay attention to IC 11 and 13):
# **first:**
## [image]

# **second:**
## [image]

agramfort, 2015-05-12:
maybe a good one for @jmontoyam ?

teonlamont, 2015-05-14:
reproduced using the sample data:

```python
path = mne.datasets.sample.data_path()
raw_file = path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_file, preload=True)

ica = mne.preprocessing.ICA(.9, method='infomax', random_state=42)
ica.fit(raw, decim=4)
p = ica.plot_components(show=True)

ica2 = mne.preprocessing.ICA(.9, method='infomax', random_state=42)
ica2.fit(raw, decim=4)
p2 = ica2.plot_components(show=True)
```

teonlamont, 2015-05-14:
@Eric89GXL @dengemann I swear I want a release *tout de suite* but this looks like something that might need to be addressed.

Eric89GXL, 2015-05-14:
I agree, this is a blocker

teonlamont, 2015-05-15:
closed for #2096

closed 2015-05-15 by teonlamont