mne-tools/mne-python #2086: saving foci for every single time sample

atalanti, 2015-05-12:
Hello everyone,

I would like to ask if it is possible to record in a movie the foci at every time step, when at source space.

If I use:

vertno_max, time_idx = stc.get_peak(hemi='lh', time_as_index=True)
brain.set_data_time_index(time_idx)
brain.add_foci(vertno_max, coords_as_verts=True, hemi='lh', color='blue', scale_factor=0.6)
brain.save_image('dSPM_map_lh5-15.png')

I get the foci as the location in which there is the maximum dspm value for all the duration of my epoch. So I get an image at the time and the location where the maximum dspm value happened.

Is there a way, when I use brain.save_movie, to record the foci at every step? I mean, at EVERY time  step while recording video, to compute and show the location of the brain with the maximum dspm value. 
So far I can record the dspm values with:

for hemi in ['lh', 'rh']:
    stc = read_stc(stc_fname % hemi)
    data = stc['data']
    times = np.arange(data.shape[1]) * stc['tstep'] + stc['tmin']
    brain.add_data(data, colormap='hot', vertices=stc['vertices'], smoothing_steps=10, time=times, time_label='%0.3f s', hemi=hemi)
brain.save_movie('example_current3.mov', time_dilation=3., tmin=tmin, tmax=tmax, framerate=24, interpolation='quadratic', codec='mpeg4')
brain.close() 

Thank you very much in advance

agramfort, 2015-05-13:
you need to do this manually ie save all frames and make a movie from the
frames

atalanti, 2015-05-13:
Alright. Thank you.

closed 2015-05-16 by Eric89GXL