mne-tools/mne-python #2087: max_pca_components no reset during preprocessing.ica.ICA._reset

mdelpozobanos, 2015-05-12:
Hi everyone,

I've found that when refitting an ICA object, the property max_pca_components is not set to its original values during reset. This may lead to unexpected behaviour. 

Example:

+ Set "max_pca_components=None", so that the object automatically sets it to use all computed PCs through the line "self.max_pca_components = len(picks)" when fitted with a dataset. 

+ Now refit the object with a dataset with different number of sensors. As max_pca_components is not reset to None, the object won't be configured to use all the computed PCs (the previous value is kept), producing unexpected behaviour.

Thank you
Marcos

PS: I'm new on github so I'm not sure how this works. Should I fix this issue by myself and request a pull?

agramfort, 2015-05-12:
sounds like a bug indeed. Can you provide a code snippet to reproduce the
problem using the sample dataset?

thanks

dengemann, 2015-05-12:
Indeed sounds like a bug. I can take a look later or assist you in doing
your first PR ;)

mdelpozobanos, 2015-05-13:
If you could give me a week, I'll pull a fix proposal :-)
Thanks

dengemann, 2015-05-13:
Let me do it now, this is rather urgent.

mdelpozobanos, 2015-05-13:
Oh, I see... If you haven't stated jet... I'm cloning the project now. I was excited with the idea of doing my first PR xD
If you've fixed it already. No worries. Next time

dengemann, 2015-05-13:
Yeah, next time! Happy ICAing and once more thanks for reporting this!

PR #2090 (2015-05-13): BUG: fix ICA resetting
Addresses and closes #2087 .
Thank @mdelpozobanos for reporting.
closed 2015-05-13 by Eric89GXL