mne-tools/mne-python #2088: Examples take too long to run

Eric89GXL, 2015-05-12:
A few of our examples take a very long time to run. This includes `plot_covariance_whitening_dspm.py` and especially `plot_spm_faces_dataset.py`. We should probably fix these before release.

dengemann, 2015-05-12:
Well it's the nature of some examples that they take longer, especially these two.

Eric89GXL, 2015-05-12:
There are two problems with it, though:

1. Users will not understand why the examples are taking so long, and might not like it. This we could probably deal with using comments.

2. It makes testing docs a big pain.

I wonder if there is some way we can get them to run faster.

agramfort, 2015-05-12:
In sklearn the very long examples are put in a different folder called
applications but it's different here...

Eric89GXL, 2015-05-12:
It makes it really difficult to test doc building when the spm-faces example takes so long. Is it okay if I tag a couple of the really slow examples such that we can do something like `make html_dev-noslow` or so?

dengemann, 2015-05-12:
Good idea!

PR #2092 (2015-05-14): ENH: Allow easier doc build testing
Closes #2088.

Need to actually run the `make html_dev-noslow` and check to see if there are other slow examples, but this shows the framework.
closed 2015-05-14 by dengemann