mne-tools/mne-python #2093: stc.plot() throws warning by default

mainakjas, 2015-05-13:
If I don't set any `clim` or `fmin` etc for that matter -- just a plain `stc.plot()`, it throws a warning saying. This is not desired behaviour.

cc @wronk

Eric89GXL, 2015-05-13:
I think we might need it to be that way for one release to change the
behavior to use percentiles.

wronk, 2015-05-13:
Before, stc.plot() would put in default fmin, fmid, fmax values, which we need to deprecate. I guess in the future we should have it implicitly use default percentiles with the `mne` colormap, but we need at least one release in between, correct?

@Eric89GXL, lmk if you want me to change anything though.

Eric89GXL, 2015-05-13:
I guess of all entries are none, we should use old behavior and give a
warning that it will change to clim auto next release.

wronk, 2015-05-13:
Sure, I'll open a PR in a few.

mainakjas, 2015-05-13:
I guess the warning is confusing for a new user who hasn't even started exploring these different `clim` options and is just looking to use the default. Now, he has to deal with 4 new parameters

Eric89GXL, 2015-05-13:
It's only one new parameter (and three old ones), they should use `clim='auto'` as suggested by what will be the new warning. Is that clear enough?

mainakjas, 2015-05-13:
okay for that as long as it's clear to the user that they don't need to do anything with the warning

Eric89GXL, 2015-05-13:
well... it will be the same as with any deprecation warning, in the sense that if they do nothing, the behavior might change on the next release

wronk, 2015-05-13:
See Clim fix #2094

closed 2015-05-16 by Eric89GXL