mne-tools/mne-python #2095: Apply a function to Epochs object

choldgraf, 2015-05-14:
So right now I am representing some PSD data with a kind of janky solution where I calculate the PSD of each trial with `compute_epochs_psd`. Then, I take the output of that and create a new `Epochs` object with frequency as the final axis rather than time. In the `info['description']` field, I put a list of the frequencies, and then any time I read this from disk I overwrite the times with the frequencies.

However, for something like this I'd also like to be able to do things like take the log, do zscoring, etc (this is true of time-based Epochs objects too, I suppose). The way that I've done this with other mne objects is to either use the `apply_func` method for objects, or to operate on the data directly (e.g. `evoked.data`).

However, with `Epochs` objects, I can't figure out how to do this. This is because the way you access epoch data is with a get_data() method. AFAIK, you can't set `preload=True`, and thus you can't change the epoch data in place.

Has anyone figured out a clean way of doing this, or am I being stupid and missing something? Alternatively, is there a reason that MNE doesn't allow one to read the full epochs object into memory?

agramfort, 2015-05-14:
if you preload the epochs the data is in epochs._data attribute

choldgraf, 2015-05-14:
Ah I see - the reason I didn't think this was possible was because the `read_epochs` function doesn't have a "preload" flag. Looking through the code, I see that it just sets this to true automatically. Thanks for the clarification.

closed 2015-05-14 by choldgraf