mne-tools/mne-python #2105: FIX? sample.data_path(force_update=True, download=True) does not seem to download the newest dataset

dengemann, 2015-05-16:
Am I missing something?

@Eric89GXL @agramfort

dengemann, 2015-05-16:
On manual download I get version 0.7, the shrunk cov files are missing.

dengemann, 2015-05-16:
Build november 2013

Eric89GXL, 2015-05-16:
The shrunk cov files haven't been added yet. It probably is re-downloading fine, and we just need to update the dataset.

Eric89GXL, 2015-05-16:
If you do it with `verbose=True` you should see the status update as it downloads to confirm it's updating.

dengemann, 2015-05-16:
> The shrunk cov files haven't been added yet. It probably is re-downloading fine, and we just need to update the dataset.

we should do that because otherwise the example will be broken on mne stable.

Eric89GXL, 2015-05-16:
For now we'll have to work with Martin or Hari to get that done. I might know more on Wednesday that will allow us to do it ourselves, which is why I've been waiting.

dengemann, 2015-05-16:
> For now we'll have to work with Martin or Hari to get that done. I might know more on Wednesday that will allow us to do it ourselves, which is why I've been waiting.

Eric89GXL, 2015-05-16:
I haven't considered it a blocker because I think we can get it working in short order, and the probability of someone hitting the issue in the couple of days there is a discrepancy should be small

dengemann, 2015-05-16:
> and the probability of someone hitting the issue in the couple of days there is a discrepancy should be small

Well, a bunch of examples won't work without the update.

Eric89GXL, 2015-05-16:
True, I think there are three or four that need it. I think the probability of someone hitting those in the intervening days is small, but if you disagree we can wait

Eric89GXL, 2015-05-16:
...and you should work with Hari and Martin about getting it included

Eric89GXL, 2015-05-16:
The file is here:

http://staff.washington.edu/larsoner/sample_audvis-shrunk-cov.fif

Eric89GXL, 2015-05-16:
Never mind, I'll take care of it

Eric89GXL, 2015-05-18:
All better

closed 2015-05-18 by Eric89GXL