mne-tools/mne-python #2108: indexing Dipole

agramfort, 2015-05-17:
it would be great to be able index instances of Dipole e.g. to do:

`dipole[dipole.gof > 0.8].plot_locations()`

we should be able to do:

```
dipole[dipole.gof > 0.8]
dipole[:5]
```

as if we had 1d numpy array of dipoles.

@jmontoyam It would simplify your scripts.

Eric89GXL, 2015-05-17:
@agramfort how urgently is this needed? It should be fairly straightforward so I could do it, but it might be good for a new contributor if they want to take it

agramfort, 2015-05-17:
I think @jmontoyam should give it a try as a first contrib.

jmontoyam, 2015-05-25:
@agramfort, @Eric89GXL 

What do you think about this solution?
https://github.com/jmontoyam/mne-python/blob/fix-issue-2108-Dipole/mne/dipole.py#L193

agramfort, 2015-05-25:
Open a PR please

Thx

PR #2171 (2015-06-01): [MRG] Fix issue 2108 dipole
fixes #2108 

taking over #2147

cc @jmontoyam
closed 2015-06-01 by Eric89GXL