mne-tools/mne-python #2109: Combining runs with different dev_head_t

teonlamont, 2015-05-18:
I have data that from two different time periods but within the same session for a subject: Experiment A, B, A'. I took marker measurements at the beginning, in between, and at the end of each experiment. Is there a way to combine Raw objects that would have a different `dev_head_t` because of movement to have them in a common reference. I thought I read somewhere on here about an issue but I could not find it.

mainakjas, 2015-05-18:
cc @yousrabk :)

Eric89GXL, 2015-05-18:
you have a few options, in order of approximate ascending quality:

1. Concatenate data and process as usual. (If it throws an error, you can "fake it" so each runs are the same.) Compute forward solutions for each run, and average them.

2. Use `mne_map_data` to remap the data. It's not implemented in mne-python yet, but @mainakjas has code to do some or all of the heavy lifting.

3. Use MaxFilter to process the runs and translate them all into the same head position. Perhaps by the end of the summer this will be doable in Python by @wronk as it is his GSoC project, but for now you have to use MaxFilter.

mainakjas, 2015-05-18:
@yousrabk has some code but she's having a rebasing nightmare. I'll see if I can take over the rebasing and hopefully you'll have something concrete @teonlamont :)

yousrabk, 2015-05-18:
Thanks @mainakjas :) 
@teonlamont I started this code sometime ago, I didn't really test it. So test it on your data, btw I find some time to finish it.

teonlamont, 2015-05-18:
cool, I will test it once it comes out of rebase purgatory. thanks @yousrabk and @mainakjas!!

mainakjas, 2015-05-19:
@yousrabk : https://github.com/mainakjas/mne-python/tree/map_dev_head

sometimes cherry-pick is better than rebase :)

yousrabk, 2015-05-19:
Thanks @mainakjas :)

teonlamont, 2015-09-29:
closing for now. someone can reopen this later

closed 2015-09-29 by teonlamont