mne-tools/mne-python #2111: BUG: combine events?

kingjr, 2015-05-18:
@teonlamont @dengemann 
I'm unsuccessfully trying to run the combine_event example: 

https://github.com/mne-tools/mne-python/blob/e6c9719a941e2dc9c39d80749d79c8349ed8896d/examples/make_event_id.py

It behaves as if @teonlamont's PR hadn't been taken into account:

```
mne-python/mne/epochs.py in __init__(self, info, event_id, tmin, tmax, baseline, picks, name, reject, flat, decim, reject_tmin, reject_tmax, detrend, add_eeg_ref, verbose)
     64         if isinstance(event_id, dict):
     65             if not all(isinstance(v, int) for v in event_id.values()):
---> 66                 raise ValueError('Event IDs must be of type integer')
     67             if not all(isinstance(k, string_types) for k in event_id):
     68                 raise ValueError('Event names must be of type str')

ValueError: Event IDs must be of type integer

```

Maybe I did something wrong, but haven't been able to find what.

teonlamont, 2015-05-18:
@kingjr my pr was closed, i'm not sure how it made it into the example, but I have been thinking about raise the issue again. there are useful cases to have it included like the case you have.

dengemann, 2015-05-18:
Ah ok, and the examples has been removed. So we can close ...

closed 2015-05-18 by dengemann

kingjr, 2015-05-18:
Ah ok, I had thought it was merged, my bad. I'll try to see whether I can make use of https://github.com/mne-tools/mne-python/pull/1990.