mne-tools/mne-python #2113: channel position in plot_topomap does not always match head drawing

alexandrebarachant, 2015-05-18:
In plot_topomap, channel positions are rescaled to match head drawing. 

The rescaling is done here :
https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L240-L245
for each axis, it takes the min and the range, and stretch the coordinates so that the range cover the entire head.

This is fine when the montage is dense and "uniform", for example this one : 
[image]

but if you want a montage with channels only on the left hemisphere, this is what you get :
[image]

I have no perfect fix in mind. a cheap solution would be to add a option to disable this scaling.

Eric89GXL, 2015-05-18:
We could have a `position_scaling='auto'` that you could set to `position_scaling='equal'` that would make it so the x- and y-scalings were equal. But that still wouldn't help with the alignment / centering issue, which is really the problem here. Maybe x=0 should always be in the center of the head, left-to-right? I wonder what that would look like. @dengemann @teonlamont I assume you use these the most, WDYT?

teonlamont, 2015-05-18:
it would be great if the scaling was based on all the channels but at that point the points have been selected. I think setting x=0 to be center is the most sensible option.

Eric89GXL, 2015-05-18:
nevermind I'm on it

agramfort, 2015-05-19:
@alexandrebarachant thanks for the bug report. Please give a try to #2114 to see how well it fixes your problem.

alexandrebarachant, 2015-05-19:
Thanks a lot.
I tried the code,

@Eric89GXL , you remove the offset cancellation (- x.min() ) so now there is a problem for channel position that have a different origin. 

It is the case for the "biosemi" layout : 
```
import numpy as np
from mne.channels import read_layout
from mne.viz import plot_topomap
layout = read_layout('biosemi')

plot_topomap(np.random.randn(64),layout.pos,vmin=-30,vmax=30)
```

[image]

I can make it works if I manually center the electrodes position on Cz

However, it will only fix partially the underlying problem.
The issue with a scaling based on min/max of the electrode position is that you assume there is an electrode in your set that is near the border of the head for each axis. This is not always the case.

I think the best solution is to give the option to the user to bypass this scaling, so if he is not happy with the default behaviour, he can always adjust the position by itself.

Eric89GXL, 2015-05-19:
So we can't assue that x=y=0 is considered "centered" on the head? That's unfortunate. I wonder if we should standardize that...

@alexandrebarachant I guess we could allow the user to specify the center and scalings for cases like this. Maybe `rescaling='auto'` should be what there is in this PR, and `rescaling=dict(center=(x, y), scale=(x_scale, y_scale)` would give the user full control. That combined with actually standardizing the layouts might be the best way to go...

Eric89GXL, 2015-05-19:
@alexandrebarachant @agramfort @teonlamont @dengemann WDYT?

agramfort, 2015-05-19:
whatever works as long as it stays simple :)

API you propose makes sense. Not sure if we can simplify.

alexandrebarachant, 2015-05-19:
@Eric89GXL x=y=0 is considered "centered" on the head, this is where the electrode Cz is located.
The problem is that there is no consensus about the origin in electrode position file. For example, in the biosemi layout provided with mne, the coordinate of Cz is ( 0.41570618,  0.51256968 ).

Since you removed the centering, if the electrode position given to the function is not referenced to the same origin, we get this shift.

I guess the best option is to come back to the original commit, and add the rescaling api you propose.
If your not in a hurry, i can make the pull request.

Eric89GXL, 2015-05-19:
@alexandrebarachant my point is that perhaps we should change the layout file so that Cz is at 0, 0 if that is indeed the standard. Then users wouldn't have to worry about setting the scaling in most cases.

agramfort, 2015-05-19:
we did not create the layouts files they are shared between packages AFAIK.
Let's avoid creating our own world ;)

PR #2114 (2015-05-21): MRG ENH: Center and allow ch_type=None
Closes #2113.
Closes #2027.

Example code:
```
import os.path as op
import mne
from mne.datasets.testing import data_path
evoked_fname = op.join(data_path(), 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(evoked_fname)[0]
evoked.pick_types(meg=False, eeg=True)
evoked.plot_topomap()
```
On current `master` this throws an error. This is what we now get on this PR:

[image]

If change the last lines to select just the left EEG channels, with `ch_type='eeg'`:
```
left_ch = [ch['ch_name'] for ch in evoked.info['chs'] if ch['loc'][0] < 0]
e [... cut]
closed 2015-05-21 by dengemann