mne-tools/mne-python #2116: ENH: add I/O for HCP MEG data

dengemann, 2015-05-19:
I think MNE-Python should support HCP data processing.
Not only is this a timely thing, but it would provide us with many opportunities for testing and validating algorithms, etc.

This is a challenge for a couple of reasons:

- Native formats are BTI
- The HCP dataset does not ship the headshapes, this makes coreg impossible
- The data provided do not allow for conversion to the neuromag format (fiducials are not in the required coordinate system)
- The source models provided are produced by the `wb_command` (the HCP workbench tool) which uses resampling instead of subsampling, as a consequence, mapping to Freesurfer surfaces is not straight-foward / impossible (?) and dirty hacks are required to make plotting work.
- Morphing and some advanced PySurfer functions, e.g, ROI to surface stuff, will not work ... #ExcrutiatingPain

For more details see the ``hcp_anatomy.m`` in the megconnectome pipeline and the HCP reference manuals. Note. We already discussed these issues for a while on the HCP mailinglist. These are real issues

Currently, what we can do is the following:

1) bypass the chain of transforms in read_bti (via a new ``convert_to_neuromag`` argument, puts identity matrices)
2) then we can use the transforms provided + the coreg. Their source models are in helmet space. So the transform would map to their model if we don't convert BTI data and we can run forward + inverse
3) write some functionality to fake freesurfer surfaces. As we cannot easily define 'nuse' et al. we can pretend to have an untessalated surface. But then we would need to re-write the subject's inflated surface files.

My vision of things would be to start with some I/O functions in ``mne.datasets``:

- ``read_trans`` based on the HCP transform ASCI files, there we notable need the ``transforms.bti2vox``
- ``read_source_spaces``, the 2d source model in a .mat file
- later some readers for connectivity results and the HCP inverse solutions:
   - at least ``read_source_estimates``, which would be useful for computing connectivity with our methods / other methods in python
- later, later data fetching functionality based on ``boto`` the Amazon S3/E2 Python library (yes, the HCP data is on Amazon)

I already started draftin some small parts of what I listed here. If you are not opposed I would start an early WIP PR.
This should be one of the priorities for the 0.10 release btw.

@Eric89GXL @mainakjas @agramfort @SherazKhan , ...

SherazKhan, 2015-05-19:
Hi Denis,

I also fond HCP data hard to work with, after loosing all hopes, now I am
working with OMEGA dataset and found it much more open.
OMEGA: The Open MEG Archive
http://www.sciencedirect.com/science/article/pii/S1053811915003183

Soon I Will share my pipleline on the mne list for it.

Best

Sheraz

> I think MNE-Python should support HCP data processing.
> Not only is this a timely thing, but it would provide us with many
> opportunities for testing and validating algorithms, etc.
>
> This is a [...]

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.

dengemann, 2015-05-19:
@SherazKhan we should also support the OMEGA datasets. I recently talked to the devs, it seems they won't have an API in the for downloading the datasets in the nearer future for downloading the data. How do you acceess the data programmaticaly? Please don't hesitate to share your code, we should add I/O and a pipeline to MNE.

SherazKhan, 2015-05-19:
OMEGA hosting LORIS is very basic, I downloaded the data manually, over
long lunch and watching Interstellar :)

> @SherazKhan we should also support the OMEGA datasets. I recently talked
> to the devs, it seems they won't have an API in the for downloading the
> datasets in the nearer future for downloading the [...]

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.

Eric89GXL, 2015-11-25:
Ideally as part of this #2170 (BTI coverage) should also be improved.

Eric89GXL, 2016-03-16:
@dengemann is this coming in the next month? If so, feel free to set the milestone back to 0.12 (I'm pushing to 0.13 now)

dengemann, 2016-03-16:
No idea how to progress on this. We would need to test all the low level bti functions. This will take a lot of time and is not necessarily effective.

Eric89GXL, 2016-03-16:
Should we close then? Or leave it open in case someone becomes motivated to
work on it?

dengemann, 2016-03-16:
We can close it I guess. Chances to find bugs there are rather low.

closed 2016-03-17 by Eric89GXL

dengemann, 2016-03-17:
Ahh sorry wrong issue i replied to. I am on it bit its now an extenal project. We will probably later taylor an API for mne-python. Now dev is taking place in mne-tools/mne-hcp.