mne-tools/mne-python #2119: GAT params consistency?

kingjr, 2015-05-19:
The parameters of the GAT are currently not 100% consistent.

Some are passed to ```__init__()``` (```clf```, ```cv```, ```train_times```, ```predict_mode```, ```n_jobs```) others are passed to ```predict()``` (```test_times```) or ```score()``` (```scorer```). 

Historically this is because there's been a tension between adopting the sklearn style (all params at init) and practicality (we often test different ```test_times``` or ```scorer``` from the same fit). 

However for sake of consistency, I think we should make an explicit decision here. I tend to favour params specification where they are relevant (```fit(epochs, y, train_times=None)```, ```predict(epochs, test_times```, ```score(epochs, y, scorer=None)```) because I find it more practical to use in my scripts.

Anyone else with an opinion?

agramfort, 2015-05-19:
>  I find it more practical to use in my scripts.

:)

think about you but also the reason which is why you integrate this into
MNE : make this tool easy to use by others.

The reason we have sklearn API is because more and more people know sklearn
and they would pick up MNE fast if we follow its conventions. That's also
why we use numpy as much as possible. You reduce the learning curve...

in summary if it's just to save 1 line I am -1 :)

kingjr, 2015-05-19:
Ok, but then you're ok to set ```test_times``` and ```scorer``` at init?

dengemann, 2015-05-19:
> Ok, but then you're ok to set test_times and scorer at init?

Yes. Sounds more sklearnish to me.

kingjr, 2015-05-19:
All right, it's settled.

kingjr, 2015-05-19:
see https://github.com/mne-tools/mne-python/pull/2120

closed 2015-05-19 by kingjr