mne-tools/mne-python #2121: Potential bug with concatenate_raws

teonlamont, 2015-05-19:
I tried to concatenate two `RawKIT` objects with `preload=True` for each and this worked fine. Then I tried `preload=False` and this gave an error when I tried `raw.preload_data()`. I get a 
`ValueError: total size of new array must be unchanged` so I delved down in the code and found in `_read_segment`, the code isn't being told where file1 is stopping and file2 is beginning, it tries to read it all from file1.

Are we supposed to be able to do `preload_data` for concatenated unpreloaded files. Also, for naming purposes, should it be called `load_data()` instead of `preload_data()` since it's not being preloaded

agramfort, 2015-05-19:
> I tried to concatenate two RawKIT objects with preload=True for each and this worked fine. Then I tried preload=False and this gave an error when I tried raw.preload_data(). I get a
> ValueError: [...]

can you reproduce the pb with the test kit file we have in the repo?

> Are we supposed to be able to do preload_data for concatenated unpreloaded files. Also, for naming purposes, should it be called load_data() instead of preload_data() since it's not being preloaded

-1 we should avoid API changes unless there is a clear API issue.

teonlamont, 2015-05-19:
```python
import mne
from mne.io.kit.tests.test_kit import sqd_path

raw = mne.io.read_raw_kit(sqd_path, preload=False, verbose=False)
raw2 = mne.io.read_raw_kit(sqd_path, preload=False, verbose=False)

mne.concatenate_raws([raw, raw2])
raw.preload_data()
```

Eric89GXL, 2015-05-19:
We also should check this for `EDF`, it probably suffers from the same issue.

agramfort, 2015-05-19:
we need a generic test for Raw objects...

teonlamont, 2015-05-19:
just confirmed that it does work for RawFIF. 

does not work for RawEDF
```
import os.path as op
import mne
from mne.io.edf.tests.test_edf import bdf_path

raw = mne.io.read_raw_edf(bdf_path, preload=False, verbose=False)
raw2 = mne.io.read_raw_edf(bdf_path, preload=False, verbose=False)

mne.concatenate_raws([raw, raw2])
raw.preload_data()
```

agramfort, 2015-05-20:
that's a release blocker to me... :(

can you have a look?

Eric89GXL, 2015-05-20:
I might have time this morning, unless @teonlamont you're on it already?

teonlamont, 2015-05-20:
@Eric89GXL feel free to take the lead. I was only able to identify where it was working in fif.raw
https://github.com/mne-tools/mne-python/blob/master/mne/io/fiff/raw.py#L389. I can't get to it at the moment.

Eric89GXL, 2015-05-20:
cool, I'm on it

PR #2127 (2015-05-27): MRG FIX: Unifying raw reading
Refactoring is only complete for `RawFIF` so far. I have moved `_read_segment` into `_BaseRaw`, and it calls `_read_segment_file`, which is a more targeted function. Now I need to transform the reading functions for KIT and EDF to work the same way. Hopefully we'll end up with more red lines than green ones.

Closes #2121.
closed 2015-05-27 by agramfort