mne-tools/mne-python #2125: no activation on source space on greater frequency bands

atalanti, 2015-05-20:
Hello everyone!

In my raw eeg data i have activation in all frequency bands. I have computed tha dSPM values for my epochs for one event for each of delta,theta, alpha and beta slow and fast frequency bands separately and they are all different from zero (1-4.5). Delta band has bigger values (up to 6) than the rest of them but the difference is small. However when i apply the inverse operator and take the source space, i can only see activity in delta and low theta band. Nothing on grater frequencies. 
Could you please explain me why this is happening? Is it because of the smaller dspm values of the greater bands? Also, in case i use mne instead of dspm i see on source space no activation at any band. 

Thank you very much.
Atalanti

agramfort, 2015-05-20:
if you use the same noise cov for all frequencies then dSPM will be much
lower in high frequencies. You can fix this scaling problem by computing
the noise cov and an inverse operator after filtering the data in the band
of interest

HTH

atalanti, 2015-05-20:
Thank you very much.
Best
Atalanti

closed 2015-05-20 by agramfort