mne-tools/mne-python #2137: Font on the stable version

teonlamont, 2015-05-22:
Is it intended to have different fonts between the stable and the development websites?

Eric89GXL, 2015-05-22:
No, they should be the same. @agramfort did something go wrong with your builds?

Eric89GXL, 2015-05-22:
Oh, wait -- yes for just right now because the stable website hasn't been updated. @agramfort can you update it?

Eric89GXL, 2015-05-22:
http://martinos.org/mne/stable/ does not reflect http://mne-tools.github.io/stable/

mainakjas, 2015-05-22:
Also, is this gap / whitespace intentional?

[image]

Eric89GXL, 2015-05-22:
No -- it might be some bug with `sphinx`

agramfort, 2015-05-22:
the white gap will be fixed later. We'll switch to sphinx-gallery soon.

Eric89GXL, 2015-05-22:
I'll change the title to be about that

Eric89GXL, 2015-05-22:
I'm also going to make a new tag "Sprint 2015" and move this and some other doc issues there

agramfort, 2015-07-14:
fixed by #2279

closed 2015-07-14 by agramfort