mne-tools/mne-python #2143: ica.plot_sources is problematic if inst.info['bads'] != []

teonlamont, 2015-05-25:
When trying to plot ica sources, I run into an error:
```python
RuntimeError: Epochs don't match fitted data: 207 channels fitted but 208 channels supplied. 
Please provide Epochs compatible with ica.ch_names
```
but that's because my `epochs.info['bads'] = ['MEG 130']`. this channel is dropped properly when computing the ICA but it's not being handled properly with the underlying `_transform_epochs`. The function should take a `picks` and if `picks=None`, remove the bads.

Thoughts @dengemann?

Eric89GXL, 2015-05-25:
likely dup of #2140

dengemann, 2015-05-25:
yes, this seesm to be a regression. I'm pretty sure this part of the code has been working.

teonlamont, 2015-05-25:
this is a bit different, it's not working for me. here's an example with the sample data:

```python
import os.path as op
import mne

raw_file = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample',
                   'sample_audvis_raw.fif')
evt_file = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample',
                   'sample_audvis_raw-eve.fif')
raw = mne.io.read_raw_fif(raw_file, preload=True)
evts = mne.read_events(evt_file)
epochs = mne.Epochs(raw, evts, 1, tmin=-.2, tmax=.6, preload=True, verbose=False)
ica = mne.preprocessing.ICA(.9, random_state=42, method='infomax')
ica.fit(epochs, decim=2)
ica.plot_sources(raw)

RuntimeError: Epochs don't match fitted data: 364 channels fitted but 365 channels supplied. 
Please provide Epochs compatible with ica.ch_names
```

dengemann, 2015-05-25:
mhmm ... 
ica.fit will ignore the bad channes by default.
I think I regularly pick bad channels at the Epochs stage why I don't have this issue.
In fact the semantics says that ica.ch_names only includes the channels used on fit.
I would not change this.
Not sure what would be a good deal.

PR #2146 (2015-05-28): ENH picks and FIX typo
closes #2143.
closed 2015-05-28 by Eric89GXL