mne-tools/mne-python #2149: FIX/ENH: clims auto method can fail with thresholded maps

dengemann, 2015-05-25:
common scenario: significant local activation after permutation clustering, all other vertices are zero, percentliles would yield

fmin=0, fmid=0, fmax=55.5

This could somehow be improved / fixed.

cc @wronk

wronk, 2015-05-27:
First thing that comes to mind is to check for zeros and exclude those from the percentile calculation. But then there will be issues if the vertex values aren't all positive. I could also ask in the params if the map has been thresholded, but that seems to clunky too. 

Do you have any good ideas as to how to accomplish this without trying to handle too many special cases?

agramfort, 2015-05-27:
removing all the zeros from the computation seems fair. It will also nicely
handle sparse estimates

dengemann, 2015-05-27:
> removing all the zeros from the computation seems fair. It will also nicely
handle sparse estimates

+1

wronk, 2015-05-27:
I'm hesitant to start throwing values out in this calculation because I can't shake the feeling that we'll fix this corner case and end up with many more unforeseen issues. I can give that a shot and see if it gives any weird issues.

Sidenote: your example issue with fmin/fmid being equal should only occur if you manually defined the percentages, right (using the `lims` key in `clims`)? If using `'auto'` then the fmid value is set as the average between `fmin` and `fmax`: 

`if clim == 'auto':`
`# Set upper and lower bound based on percent, and get average between`
`ctrl_pts = np.percentile(np.abs(stc_data), [96, 99.95])`
`ctrl_pts = np.insert(ctrl_pts, 1, np.average(ctrl_pts))`

Would simply entering new percentiles in the `lims` key not fix that?

dengemann, 2015-05-28:
> your example issue with fmin/fmid being equal should only occur if you manually defined the percentages, right (using the lims key in clims)? If using 'auto' then the fmid value is set as the average between fmin and fmax:

it also fails with 'auto' if this is your question.

Maybe we could have some conditional that checks if there are values that are strictly zero. If this is the case use non-zero values, else all.
That would simplify to using non-zero values ;)

agramfort, 2015-05-28:
true zeros never happen by chance. They are meant to be ignored from
visualization.

Eric89GXL, 2015-05-28:
@wronk I can take care of this one if you want, I'd rather have your brain power going toward the GSoC projects :)

dengemann, 2015-05-28:
@Eric89GXL you're max-filtering @wronk <https://github.com/wronk> --
removing external nerd-sniping fields ;)

wronk, 2015-05-28:
Haha, sure.

Will the `auto` limits change if you detect thresholding? Say you threshold and find 100 vertices that make some arbitrary significance cutoff. If you use percentiles, then only 4 of them will be at or above `fmid` now.

dengemann, 2015-05-28:
@wronk I think it's not relevant if you don't have zeros in your data. If you have zeros in your data this because you applied a mask. Then using non-zero values is what you want.

wronk, 2015-05-28:
Ya, okay. A mask would have to be applied pretty deliberately, so that makes sense.

dengemann, 2015-05-28:
This is what we do in our examples.

PR #2168 (2015-06-01): MRG: Fix limits
Closes #2149.

Turns out that excluding zero points actually breaks a lot of examples, and the current implementation seems to work pretty well while simplifying some examples. Ready for review/merge from my end.
closed 2015-06-01 by agramfort