mne-tools/mne-python #2157: FIX: backport raw.plot proj toggle to 0.9/maint

dengemann, 2015-05-28:
How to reproduce?
With sample data plot with show_options=True then move up and down through channel groups and toggle projs.

```
Error in sys.excepthook:
Traceback (most recent call last):
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/IPython/core/application.py", line 197, in excepthook
    return self.crash_handler(etype, evalue, tb)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/IPython/core/crashhandler.py", line 180, in __call__
    input("Hit <Enter> to quit (your terminal may close):")
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/IPython/utils/py3compat.py", line 150, in input
    return builtin_mod.raw_input(prompt)
RuntimeError: can't re-enter readline

Original exception was:
Traceback (most recent call last):
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/backend_bases.py", line 1870, in button_press_event
    self.callbacks.process(s, mouseevent)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/cbook.py", line 540, in process
    proxy(*args, **kwargs)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/cbook.py", line 415, in __call__
    return mtd(*args, **kwargs)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/widgets.py", line 558, in _clicked
    if (t.get_window_extent().contains(event.x, event.y) or
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/text.py", line 741, in get_window_extent
    bbox, info, descent = self._get_layout(self._renderer)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/text.py", line 311, in _get_layout
    ismath=False)
  File "/Users/dengemann/anaconda/lib/python2.7/site-packages/matplotlib/backends/backend_macosx.py", line 166, in get_text_width_height_descent
    six.text_type(s), family, size, weight, style)
RuntimeError: CGContextRef is NULL
```

Eric89GXL, 2015-05-28:
`matplotlib`'s problem:

https://github.com/matplotlib/matplotlib/issues/2654

Eric89GXL, 2015-05-28:
Oh boy, the bug is a year and a half old...

Eric89GXL, 2015-05-28:
We could try what seaborn did and use `fig.set_tight_layout` as well for the tight layout code:

https://github.com/mwaskom/seaborn/pull/546/files

agramfort, 2015-05-28:
sounds like a plan...

closed 2015-05-30 by agramfort

dengemann, 2015-05-30:
> Closed #2157 <https://github.com/mne-tools/mne-python/issues/2157> via

Thanks Eric for taking care of this!

PR #2166 (2015-05-30): MRG: Fix OSX plotting
Closes #2157.

~~I need to test to see if this actually works, but it follows suggestions in https://github.com/matplotlib/matplotlib/issues/1852. I have a feeling it will not fix our immediate problem.~~