mne-tools/mne-python #2160: Inconsistency in head positions (translations) reported by maxfilter vs. get_chpi_positions

jhouck, 2015-05-28:
The translations I get from get_chpi_positions differ considerably from what I see in the maxfilter gui.  The last sample in the maxfilter gui is 352.013s, where yaw is 9.8, roll is -1.6, and pitch is -17.s.  Translation in the x, y, and z directions respectively is -1, -3, and -9 mm. 

The output I get from get_chpi_positions differs somewhat (see https://gist.github.com/jhouck/d7cbf397c2c155811383 for the code).  Using the text position file (available from https://www.dropbox.com/s/al5eas8ike72vn9/M87110696_rst_raw.pos?dl=0), the last sample is at 352 ms, where yaw is 9.82, roll is -1.35, and pitch is -17.27 degrees.  This looks pretty good.  Translation in the x, y, and z directions, however, is .67, 5.5, and 47.74 mm, which is less close.  This looks suspiciously like a head origin, and when I do a show_fiff -vt 222, this is really very close to the origin there (1.05, 5.82, 47.72).  Is there an additional transformation I should be doing?  

Unfortunately if I use mne_process_raw to save out a chunk of the file, maxfilter no longer buys it as a raw file.  The complete file can be downloaded from https://www.dropbox.com/s/p0h5umvwlzqb4iy/M87110696_rst_raw.fif.gz?dl=0 . The file is relatively unprocessed; I used maxfilter only to extract the head positions and downsample.  The full command line was /neuro/bin/util/maxfilter -f M87110696+Study20100201+rst_raw.fif -o M87110696_rst_raw.fif -autobad off -format float  -hpicons -headpos -hp M87110696_rst_raw.pos -ds 2 -force -v

dengemann, 2015-05-28:
Jon, is this as suspected what drives your rank values?

Eric89GXL, 2015-05-28:
Are you getting the positions from a text file or from a raw instance? If it's the latter, what `t=0` means might differ. Maxfilter probably doesn't subtract off the "initial skip" whereas mne-python does. You can figure out the time difference from `raw.first_samp / raw.info['sfreq']`. See the Notes section here:

http://martinos.org/mne/stable/generated/mne.get_chpi_positions.html#mne.get_chpi_positions

jhouck, 2015-05-28:
@dengemann It's a candidate; the idea was to examine each maxfilter processing step separately to see where things broke down.  And they broke down immediately. :) At this stage nothing much has happened to the file and the rank is probably accurate as reported by raw.estimate_rank as 306.

jhouck, 2015-05-28:
@Eric89GXL Unfortunately the results are about the same, although the text file rounds off after 2 digits.  Using the last sample in the raw file the translations are 0.66559356,  5.50761972, 47.74046205; using the last sample of the text file they are 0.67,   5.51,  47.74.

Eric89GXL, 2015-05-28:
I don't get it, aren't those the same?

jhouck, 2015-05-28:
They're the same as each other, not the same as the  -1, -3, and -9 reported by the maxfilter gui.

dgwakeman, 2015-05-28:
I believe the gui is just doing an annoying subtraction from the initial head position. Check if the initial value close to  [0 0 0]. The numbers the gui says are shown using the same data as maxfilter produces in the text file output. guis suck...

jhouck, 2015-05-28:
@dgwakeman You must be right -- the manual says that "The x, y and z coordinates [ mm ] represent positions of the SSS expansion origin (r0) in respect to the device coordinate point (0, 0, 0)" (p.49). In this case no SSS was performed but it did the subtraction anyway.

By the way, is there any SSS-processed sample data available?  I'd seen a reference to it in an old issue, but I didn't see it in the MNE sample data I re-downloaded last night.  It would be a handy way to detect user error for things like this.  Or totally-justifiable user confusion, as the case may be.

dgwakeman, 2015-05-28:
Hmm. actually, I think the text you quoted means that it should be relative to device coordinates not the initial fit point (though, in my mind it is likely a documentation bug of neuromag's: I would recommend emailing neuromag). After checking the manual's (2.0, 2006, newest version I have) gui section I see the "log window" figure, which has shockingly small values given the text you read in your manual. I believe this evidence supports the hypothesis that this is a Neuromag documentation bug, and that the "log window" shows the position relative to the position at the point of the initial head estimation.

jhouck, 2015-05-29:
These raw files also include the initial HPI measurement. Would this also be a contributing factor?  All of my raw data with cHPI were collected this way.

Eric89GXL, 2015-05-29:
I'm not sure the best way to figure out how Elekta is calculating those values in their program. At this point I don't think the problem is at our end, since both reading from the raw channels and reading from the `.txt`/`.pos` file given by Maxfilter produces the same result. So this seems to be an Elekta software mystery, where they are doing something other than reporting those values in whatever status element you're looking at. Like @dgwakeman said they might be doing deviations relative to the initial head position in head or device coordinates, but it's hard to know without documentation of that feature. I guess you could do a bunch of calculations and figure out which ones are correct, but at least in the meantime I don't think you need to worry about the correctness of the values.

jhouck, 2015-05-29:
True -- as long as the relative deviations over time are accurate, which they appear to be, the absolute position doesn't really make a difference.

Eric89GXL, 2015-12-12:
I'll go ahead and close this, but if it seems to cause problems feel free to reopen

closed 2015-12-12 by Eric89GXL