mne-tools/mne-python #2161: Discrepancy between docstring and code for GeneralizationAcrossTime

choldgraf, 2015-05-28:
Hey all - I was looking into the default behavior of the GAT class, and there seems to be a difference between the docstring and the code for when the user does not provide their own fitting class:

From the docstring:

    clf : object | None
        An estimator compliant with the scikit-learn API (fit & predict).
        If None the classifier will be a standard pipeline including
        StandardScaler and a linear SVM with default parameters.

The code:

        if clf is None:
            scaler = StandardScaler()
            estimator = LogisticRegression()
            clf = Pipeline([('scaler', scaler), ('estimator', estimator)])

Shouldn't `estimator` be an instance of `LinearSVC`?

choldgraf, 2015-05-28:
(I am happy to fix with a quick PR but let me know if the docstring should read LogisticRegression, or if the code should be changed to LinearSVC)

agramfort, 2015-05-28:
Indeed PR welcome

choldgraf, 2015-05-28:
So which would you like? Should the docstring match the code, or the code match the docstring? I don't know what the original plan was.

agramfort, 2015-05-28:
Update docstring

choldgraf, 2015-05-28:
Check out #2164

closed 2015-05-29 by agramfort