mne-tools/mne-python #2162: no vertices for label

atalanti, 2015-05-28:
Dear mne-python users,

I am trying to compute the connectivity with pli at the source space. My epochs regard the response to an auditory stimuli. As a label file I am using the one of fsaverage which contains 68 labels. My source space is made with oct-6.
When I run the code (which like is the one in the example plot_mne_inverse_label_connectivity.py)

I take the following error:
Traceback (most recent call last):
  File "connect.py", line 84, in <module>
    return_generator=True)
  File "<string>", line 2, in extract_label_time_course
  File "/usr/local/lib/python2.7/dist-packages/mne-0.9.dev0-py2.7.egg/mne/utils.py", line 541, in verbose
    return function(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/mne-0.9.dev0-py2.7.egg/mne/source_estimate.py", line 2949, in extract_label_time_course
    label_tc = list(label_tc)
  File "/usr/local/lib/python2.7/dist-packages/mne-0.9.dev0-py2.7.egg/mne/source_estimate.py", line 2811, in _gen_extract_label_time_course
    raise ValueError(msg)
ValueError: source space does not contain any vertices for label frontalpole-lh

When I make a new source space with ico-5 and reproduce my fwd file I take the same error.  
Is this happening because of my data or the source space that I am using? Should I exclude this specific label from the fsaverage's label file? 
Could you please suggest me a solution or tell me what I do wrong?

Thank you in advance,
Best
Atalanti

dgwakeman, 2015-05-28:
Hi Atalanti,

This is probably a better question for the mne_analysis mailing list. But just to get you started, it sounds like you are trying to use a label on fsaverage on an individual's source space. These will not match. You should either morph the data to fsaverage, or morph the label to the individual.

HTH
d

Eric89GXL, 2015-05-28:
Yeah, you have to use labels for the individual subject. You can morph them from fsaverage to your subject if you want, but it would be cleaner to use `read_labels_from_aparc` for the specific subject instead. I'm going to go ahead and close this, feel free to reopen if that doesn't solve your problem.

closed 2015-05-28 by Eric89GXL