mne-tools/mne-python #2163: API for mixed source space inverses

haribharadwaj, 2015-05-28:
So with Alan's work, we now have support for mixed source spaces. The forward modeling once you have a mixed source space seems straightforward (already in place). However, with the inverse, there are some use cases that need to be sorted out: Ideally, orientation constraints should be supported for the cortical surface source space and loose solutions for the deep structure(s). What should the API be?

Thoughts?

agramfort, 2015-05-28:
Indeed 

Dict for loose param

Eric89GXL, 2015-05-28:
we could have it be a list the same length of the source spaces, that way if you wanted free for one volume and loose or fixed for another you could... but maybe that's YAGNI

Eric89GXL, 2015-05-28:
in which case I guess the entries would be `surf`, `volume`, and `discrete` for the three types of source spaces? I guess we could always also add list support later.

agramfort, 2015-05-28:
Yes

Eric89GXL, 2015-05-28:
@haribharadwaj you up for implementing?

haribharadwaj, 2015-05-29:
Yes.. I'll start playing with this :).. I have been separately trying to figure out the differences between gradiometers and magnetometers in terms of subcortical sensitivity.. I'm sure I'll ping you guys again with questions soon.

agramfort, 2015-05-29:
sounds great !

PR #8063 (2020-07-30): MRG, ENH: Add loose-as-dict for mixed source spaces
Closes #2163.

- [x] Check that it actually produces something reasonable
- [x] Add tests for bad inputs
- [x] Add tests that orientation priors are correct
- [x] Add tests for fixed surface + free vol (no idea if it works properly currently as some priors are actually zeroed out)
- [ ] ~~Maybe add some viz of volume?~~
closed 2020-07-30 by larsoner