mne-tools/mne-python #2170: FIX: io.bti coverage

Eric89GXL, 2015-05-31:
@teonlamont any ideas on how to improve this?

https://coveralls.io/builds/2689771/source?filename=mne%2Fio%2Fbti%2Fbti.py

For example, `_read_ch_config` is about 50 lines (including comments) and isn't tested at all.

Do you know of any small files we could add to the testing dataset that would improve our coverage? It looks like a MAGNES2500 file might help?

There is also a section with `if False:  # XXX reminds us to support this as we go` can you elaborate on that, and preferably comment out those lines instead of using `if False`?

dengemann, 2015-05-31:
I can look into this next week, I know this code well ;)
We have a config file there that  can be used.

Eric89GXL, 2015-09-16:
@dengemann if you can provide the files, I can probably make the tests for this. Can it be tested against the C tools?

dengemann, 2015-09-16:
@Eric89GXL the C tools are incomplete, this converter is special. We test it against the small test files. Btw. please don't touch I'm currently working on this code.

Eric89GXL, 2015-09-16:
Cool, just wanted to make sure someone was on this for 0.10

Eric89GXL, 2016-03-16:
@dengemann any ideas on this one? It's been pushed back quite a while

Eric89GXL, 2016-03-17:
@dengemann do you have a small file we can add to the `testing` dataset to make our coverage here better?

dengemann, 2016-03-17:
We should have a small bti files in our testing folder.

Eric89GXL, 2016-03-17:
I don't think we have the proper config file to hit the lines of the
private function mentioned above. You previously mentioned having such
files...?

dengemann, 2016-03-17:
We have a config there too. The problem might be that we don't have all
files needed to write all tests. I cannot fix this currently. We'd need
some active BTI sites with someone producing a bunch of files.

PR #3962 (2017-02-20): MRG: Fix BTi reading for Magnes2500
@dengemann this touches some of your code from 2013 (!) that apparently has a bug. I guess these lines weren't covered or ever used before this :(

I am trying to read some BTi data, I think from MAGNES2500. I hit this bug where `d` was used instead of `n`, but after fixing it I hit a bug when it tries to read the next "block", so I suspect there is something wrong with these 2500-specific lines.

Do you have some MAGNES2500 data that you were using to test/develop originally? What did you use to figure out these lines?

Closes #2170.
closed 2017-02-20 by dengemann