mne-tools/mne-python #2175: Q: `epochs_` in epochs name

Eric89GXL, 2015-06-01:
I used to create epochs from a bunch of event types, then do multiple manipulations on them ending with:
```
evoked = epochs['condition_1'].average()
```
This used to give me an evoked instance where `evoked.name == 'condition_1'`, but now I get `evoked.name == 'epochs_condition_1'`. This seems like a backward-incompatible change, but somehow it has been around for over a year (`git blame` around line 1280 in `epochs.py` traces it back to @agramfort in January 2014). Is this intended behavior, or a bug? Should I just manually get rid of the `epochs_` prefix for my use case?

agramfort, 2015-06-02:
looks like a bug to me. This epochs_ should not be there :(

Eric89GXL, 2015-06-02:
Do you remember the logic you were going for? It basically sets the name of the selected object based on whether or not the orginal object had a name != 'Unknown' (but not incorporating the original name at all). I'll go ahead and fix it. Should we backport to 0.9?

closed 2015-06-02 by agramfort

agramfort, 2015-06-03:
yes please backport.

Eric89GXL, 2015-06-03:
@agramfort done

PR #2179 (2015-06-02): MRG: Fix epoch naming
Closes #2175. Ready for review/merge from my end. I'll also backport to 0.9 once merged.