mne-tools/mne-python #2180: FIX: saving RawEDF

leggitta, 2015-06-02:
Trying to use the save method in a RawEDF object creates the following error ...
[image]

teonlamont, 2015-06-02:
if you debug here, what's the value of stim_channel?

leggitta, 2015-06-02:
For me its 72

Eric89GXL, 2015-06-02:
Can you upload a sample file somewhere for us to try it, and give a minimal script?

leggitta, 2015-06-02:
Sure. The files I'm analysing are about 1 GB each, but I'll find a smaller one to work with.

leggitta, 2015-06-02:
I've uploaded a sample .bdf with a short code that reproduces the error to the following repo

https://github.com/leggitta/SampleBDF

teonlamont, 2015-06-02:
@leggitta it seems to work with you preload the data. with the stim_channel, by default, it automatically detects the channel, which is the Status so you don't need to specify it.

@Eric89GXL do you have any clue why save breaks with `preload=False`

Eric89GXL, 2015-06-03:
No, I don't know why that would break it. I guess our tests aren't comprehensive enough.

agramfort, 2015-06-03:
:( :(

leggitta, 2015-06-03:
@teonlamont Thanks! I hadn't tried using ```preload=True```. Initially,

leggitta, 2015-06-03:
sorry, accidentally hit close and comment. 

Initially, I was worried about memory since I'm working with large files (I was trying to read them as .bdfs, crop them, and then save the smaller chunks as .fif files), but it turns out I have enough memory on my machine to read each file using ```preload=True```.

agramfort, 2015-06-03:
so there is only a pb with preload=False?

leggitta, 2015-06-03:
yes. I also tried a ```preload=tmp.txt``` and it worked

agramfort, 2015-06-03:
did someone reproduce?

Eric89GXL, 2015-06-03:
I can try it

Eric89GXL, 2015-06-03:
Reproduced and fixed (I think). @leggitta can you see if #2184 fixes it?

leggitta, 2015-06-03:
Yes, that fixed it for me.

Eric89GXL, 2015-06-03:
Thanks for the quick check @leggitta

PR #2184 (2015-06-04): MRG: Fix repr
Closes #2183.
Closes #2180.
closed 2015-06-04 by agramfort