mne-tools/mne-python #2185: legend as a subplot

teonlamont, 2015-06-04:
I was noticing in the new tutorial (thanks @dengemann) that we have a rather large legend. I was wondering if we should make an option for the legend to be a subplot of the figure, or a separate figure. That would be ideal when using figures in papers or posters to be able to retrieve separate handle

agramfort, 2015-06-04:
can you provide a pointer to what your talking about?

teonlamont, 2015-06-04:
http://martinos.org/mne/stable/auto_examples/io/plot_epochs_to_data_frame.html

[image]

Eric89GXL, 2015-06-04:
We should get rid of the legend. With that many lines it is not
particularly useful.

dengemann, 2015-06-04:
This is not a good example, it does not use any mne viz API.

teonlamont, 2015-07-24:
closing because issue was a non-MNE related problem. all and all, the legends of our plots are placed appropriately.

closed 2015-07-24 by teonlamont