mne-tools/mne-python #2187: plot_topo image size too small

ImmanuelSamuel, 2015-06-05:
[image]

ImmanuelSamuel, 2015-06-05:
Right now using this to make it large. Is there a better way?
fig = plt.gcf()
fig.set_size_inches(50,50)

Its still not nice
[image]

jona-sassenhagen, 2015-06-06:
If you can wait a bit - I'm working on improvements to plot_topo that include better determination of scale.

For now, you want to play around with the layout_scale parameter.

ImmanuelSamuel, 2015-06-06:
That is awesome! Looking forward to it.

Eric89GXL, 2017-02-16:
I assume @jona-sassenhagen has fixed this now, but @ImmanuelSamuel feel free to reopen if not

closed 2017-02-16 by Eric89GXL