mne-tools/mne-python #2188: EHN: GAT optimization

kingjr, 2015-06-05:
@dengemann @agramfort 

I'm trying to increase the speed of the GAT, and thought of initializing each classifier with the coef found in the preceeding time sample. The data is highly correlated from one clf to the next, so you could gain a lot of CPU time.

This would be potentially feasible with SGDClassifier, thanks to the `warm_start` option. However, @agramfort mentioned that it's one of the classifiers for which you have to tweak the learning rate.

Any idea on how we could proceed?

agramfort, 2015-06-06:
try using warm_start in LogisticRegression using the l-bfgs solver.

SGD is too hard to use to be trusted blindly

kingjr, 2015-06-06:
But I don't see a warm_start option for LogisticRegression?
http://scikit-learn.org/dev/modules/generated/sklearn.linear_model.LogisticRegression.html

agramfort, 2015-06-06:
arrfff indeed. It's coming with the SAG branch...

try l-bfgs anyway for speed.

dengemann, 2015-06-07:
What about the vectorization idea? By concatenating all the tests we could
call predict only n_train times, no? I would suspect speedups. But maybe
some lineprofiling before getting the hands dirty.

kingjr, 2015-06-07:
Sure, but I expect much more impact of a warm_start option than a vectorization, so I'd prioritize the warm_start.

dengemann, 2015-06-07:
Ok, let's keep all ideas and attack them in the order of relevance.

agramfort, 2015-06-07:
+1

also what does profiling says?

jona-sassenhagen, 2015-11-26:
ping @kingjr, it seems sklearn 0.17 has warm_start for logistic regression. So if this were to be included, there'd be a check for sklearn version, and if it's > .17, you'd update the args with warm_start=True?
Or if not for defaults, one of the GAT examples should be update with a note to comment out that arg if on sklearn  < 0.17?

Mayyyyybe I could even try this myself (although I'm not sure how well I understand the GAT code).

kingjr, 2015-11-26:
Jona, I had tested the warm start this summer, it's much faster. But not
all classifiers had it, so we'll need to add a series of checks to
implement it (at the moment, the estimators are cloned which doesn't help).

@dengeman we should meet for half a day to implement the new hacks on the
gat as they re likely to be even more critical.

jona-sassenhagen, 2015-11-26:
Cool stuff.

kingjr, 2016-03-04:
@kaichogami This issue (adding a warm_start option) is more difficult, but it could be a nice challenge if you like optimization.

On the TODO list:
* easy: add `warm_start` param at init
* medium: check whether `clf` can have a warm_start: this may be tricky as `clf` can be a pipeline where only the last step possesses a warm_start.
* hard: `clf` is currently cloned before each `fit` of each training time, which prevents reusing the same estimator. We'd need to change the `clf` cloning behavior, so that it only clones for each parallelization chunk, but not each training time. Then only we could use a `warm_start`
* medium: ensure that all estimators are saved independently anyway.

kaichogami, 2016-03-04:
@kingjr This sounds like nice issue to work on. Although since its difficult, I'll consume more time, I hope thats alright. I'll ping you once I have something substantial to present.
Thank you!

dengemann, 2016-03-04:
@kaichogami I'd suggest you to go for the easy subjects for now. The idea is that you get to know our community and see how we do things together. More challenges later ;)

kaichogami, 2016-03-04:
@dengemann Alright, if you say so. I'll get my hands dirty on some easier tasks! :)

dengemann, 2016-03-04:
Well if you feel bored you know where the challenges are waiting for you :)

Akshay0724, 2017-03-13:
I think this is a cool project to work under GSOC this year.

kingjr, 2017-03-13:
I'll be refactoring this class during the GSoC, so I don't think it'll be a
good choice unfortunately.

Akshay0724, 2017-03-13:
OK then, but can you please help me in pointing some of issue / or any
module. Specially I wants to know what is the present requirement of
decoding module.

Thanks

kingjr, 2017-03-13:
This will be heavily discussed in our next coding sprint in 2 weeks. It's
hard to tell before hands.

agramfort, 2017-07-22:
I think we can close now. We have a way better decoding code now.

closed 2017-07-22 by agramfort