mne-tools/mne-python #2192: Averaging doesn't work automatically after using ica.get_sources

teonlamont, 2015-06-07:
I don't know if we want to do anything about this but I noticed this behavior:

If you only have misc channels, which happens after using `ica.get_sources()`, you can't use the average method without specifying the picks. 

```python
In [10]: ics = ica.get_sources(epochs)
In [11]: e = ics.average()

---------------------------------------------------------------------------

ValueError                                Traceback (most recent call last)

<ipython-input-11-b1d5e3014cb3> in <module>()

----> 1 e = ics.average()

/Applications/packages/mne-python/mne/epochs.py in average(self, picks)

    383         """
    384 
--> 385         return self._compute_mean_or_stderr(picks, 'ave')
    386 
    387     def standard_error(self, picks=None):

/Applications/packages/mne-python/mne/epochs.py in _compute_mean_or_stderr(self, picks, mode)

    458 
    459         ch_names = [evoked.ch_names[p] for p in picks]
--> 460         evoked.pick_channels(ch_names)
    461 
    462         if len(evoked.info['ch_names']) == 0:

/Applications/packages/mne-python/mne/channels/channels.pyc in pick_channels(self, ch_names, copy)

    403 
    404         idx = [inst.ch_names.index(c) for c in ch_names if c in inst.ch_names]
--> 405         inst._pick_drop_channels(idx)
    406 
    407         return inst

/Applications/packages/mne-python/mne/channels/channels.pyc in _pick_drop_channels(self, idx)

    452             self._cals = self._cals[idx]
    453 
--> 454         self.info = pick_info(self.info, idx, copy=False)
    455 
    456         if inst_has('_projector'):

/Applications/packages/mne-python/mne/io/pick.pyc in pick_info(info, sel, copy)

    295 
    296     if len(sel) == 0:
--> 297         raise ValueError('No channels match the selection.')
    298 
    299     info['chs'] = [info['chs'][k] for k in sel]

ValueError: No channels match the selection.
```

but it does if you pick the channels

```python
In [12]: picks = mne.pick_types(ics.info, misc=True)
In [13]: ics.average(picks=picks)
```

dengemann, 2015-06-07:
But this is consistent with the documentayion of epochs.average. It has been like this all the time.

teonlamont, 2015-06-07:
ah, i see. makes sense.

closed 2015-06-07 by teonlamont