mne-tools/mne-python #2198: ENH: apply function to raw/epoch plots

choldgraf, 2015-06-11:
Following up from the epochs data viz PR. Do you guys think it would be useful to have an option to supply a function to the plot call, and then have a button/key that calls this function on the data and then re-plots? The function would have to take the currently-displayed block of data and return an array w/ the same shape and data type as what is currently displayed.

An example of where this would be useful is with high gamma in ECoG. Often it's useful to look at the raw ecog data, and if you see something noisy it can help to look at a bandpass + hilbert transformed version of the data for high gamma. I know some labs have internal code that lets you quickly switch back and forth between HG and the raw traces, and it is useful. Perhaps this would be useful to other people for other functions as well?

If it's just a keypress rather than a button, then perhaps it can be implemented without changing any of the user-facing things unless they look for it.

mainakjas, 2015-06-12:
would a text-box for changing low-pass and high-pass cutoff do it? Or do you want to apply advanced functions like hilbert ? How often is this useful ?

choldgraf, 2015-06-12:
Perhaps low-pass / high-pass would suffice. I was just imagining having a
flag like "apply_func", where this could be any function that takes a
2-d/3-d (if epochs) array and returns an array of the same shape and data
type. So whatever data was currently plotted would be passed to this
function and the currently-plotted data would be overwritten. It know many
ecog people who do this kind of thing (again, usually for high gamma), but
perhaps it is not widely applicable outside of this? Do MEG/EEG folks often
look at sub-bands of their raw data?

Eric89GXL, 2015-06-15:
I don't mind having an additional argument to do this sort of thing. I do wonder how often it would be used, though. @choldgraf how widespread is such switching functionality?

choldgraf, 2015-06-15:
Well my knowledge is largely isolated to ecog, but it's fairly useful there. The main reason for this is that ecog is a super noisy kind of data, especially if you get it from different hospitals etc. You often get weird noisiness in the raw traces, and you often wonder "Hmm, does this affect sub-band XXX that I care about?". For this reason I know ecog labs have implemented internal code that quickly switches between the raw trace and "high gamma" amplitude, since everyone loves that feature of ecog ATM. I imagined implementing this feature with an arbitrary function just because why not, so long as the user knows YMMV depending on how computer-intensive their function is...

mmagnuski, 2015-06-15:
I often find it very useful to check component contributions to the data by switching between signal with and without a specific component or set of components (because you visually experience the changes vividly as movement). I mostly do this kind of things (preprocessing) in matlab, but if I were to switch completely to python I would miss this ability. 
However to have smooth switching I usually have two datasets that I browse through (one with and one without specific component(s)) and switch between them with a button press.

agramfort, 2015-06-16:
are you taking about interactive ica component selection for raw data?

mmagnuski, 2015-06-16:
Yes, but I do it on epoched data mostly. I mention it here because it fits the general idea of applying a transformation to the visible data - and from my point of view is very useful in classifying components.

agramfort, 2015-06-17:
maybe we can use the new plot_epochs to plot ica sources as epochs?

teonlamont, 2015-06-17:
+1 for ica sources using the new plotting for ICA.plot_sources

agramfort, 2017-07-22:
still relevant?

larsoner, 2019-08-20:
I'm going to close since I think this might be better solved at the matplotlib-callback level with just a few lines. It would be cool to have an example of how to do it at some point if someone is motivated, though

closed 2019-08-20 by larsoner