mne-tools/mne-python #2200: Applying projs to Epochs made from read_epochs

teonlamont, 2015-06-15:
currently I am getting an error when I try to apply a projection to an Epochs that was created from `mne.read_epochs`. Here's how you can recreate the error with the sample data:

```python
import mne
from mne.datasets import sample
from mne.utils import _TempDir

data_path = sample.data_path()
temp_path = _TempDir()
fname_event = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
fname_raw = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
fname_epo = temp_path + 'sample_audvis_filt-0-40_raw.fif'
fname_proj = data_path + '/MEG/sample/sample_audvis_eog_proj.fif'

event_id, tmin, tmax = 1, -0.2, 0.5

# Load data
raw = mne.io.read_raw_fif(fname_raw)
events = mne.read_events(fname_event)
proj = mne.read_proj(fname_proj)

# Add a bad channel
raw.info['bads'] += ['EEG 053']  # bads + 1 more

# pick MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False, eog=True,
                       exclude='bads')
# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(mag=4e-12, grad=4000e-13,
                                                    eog=150e-6))
# This has no problem
epochs.copy().add_proj(proj).apply_proj()
# Save without proj added
epochs.save(fname_epo)

epochs_s = mne.read_epochs(fname_epo)
epochs_s.add_proj(proj)
# This raises an error
epochs_s.apply_proj()
```

agramfort, 2015-06-15:
indeed it's broken ... really broken...

read_epochs does not set detrend attribute  but also not the
_raw_times attribute.

see #2202

agramfort, 2015-06-15:
how about read_epochs returns an EpochsArray ?

dengemann, 2015-06-15:
> how about read_epochs returns an EpochsArray ?

something feels wrong about this. You would expect to read what you saved, not?

agramfort, 2015-06-15:
yeah but it's broken now. There are too many attributes that relate to raw
which we don't save. This needs to be fixed. I have no time for this these
days :(

Eric89GXL, 2015-06-15:
IMO this is another reason we need to refactor the Epochs classes.

Eric89GXL, 2015-06-15:
(I hope to get to it this week)

agramfort, 2015-06-15:
+1000

dengemann, 2015-06-15:
> IMO this is another reason we need to refactor the Epochs classes.

Seems Epochs needs some therary sessions indeed ... I think it's suffering from some developmental disorder.

PR #2211 (2015-06-27): MRG: Refactor Epochs classes
Not working yet. I hope to keep this more red than green. To-do list:

* [x] Move methods to `_BaseEpochs`
* [x] Make `EpochsArray` subclass `_BaseEpochs` instead of `Epochs`
* [x] Refactor how projection is done (move to `_BaseEpochs` constructor; #2200)
* [x] Refactor how decimation is done (add method, call it in constructor; #2178)
* [x] Refactor how rejection/dropping is done (add method; #1893)
* [x] Get EpochsArray working again
* [x] Fix RtEpochs
* [x] Fix EpochsKit, make it subclass `_BaseEpochs` instead of `EpochsArray` (which in turn had been subclassing [... cut]
closed 2015-06-27 by Eric89GXL