mne-tools/mne-python #2205: Some minor epochs.plot issues

teonlamont, 2015-06-16:
I just noticed a few cosmetic things with epochs.plot that might need to be looked at.

- [ ] Default title appears for raw.plot but not epochs.plot.
- [ ] The font size for the title of the epochs plot is very small.
- [ ] There is no title argument.
- [ ] The trigger values at the top kind of seem a bit out of place. Instead of having a row at the top, maybe we could put the trigger values next to the epoch number on the bottom row in parentheses.
- [ ] Scaling is not as convenient as we would like. Although it seems like an intuitive thing that it is -/+ for scaling, it becomes a bit cumbersome to shift for plus and not shift for minus, maybe -/= instead.

jaeilepp, 2015-06-16:
There is a fig_title argument and default title is the epochs.name.
The smaller font was kind of an ugly workaround for the time being. The layout configuration shares code with raw.plot and it needs a remake in my opinion, so i decided to leave it for following PRs. By placing the event numbers to bottom of the figure would fix this layout problem though. Should we do that instead?

teonlamont, 2015-06-16:
is there a way to rename the keyword `title` instead of `fig_title`, I wouldn't expect the current given the raw.plot arg and it would be nice to maintain consistency in keyword arguments across the package

teonlamont, 2015-06-16:
yeah, a reconfiguration of the raw and epochs plots would be welcome. I think that the trigger value should go down below the plot and be parenthetical. i don't think that it needs to have its own line

closed 2015-06-24 by dengemann