mne-tools/mne-python #2207: ENH: full size viewing mode

dengemann, 2015-06-16:
it would be nice for the browse viz functions to have an options for maximizing the window.
For reviewing in block = True mode this is very handy as it saves you one (annoying) step.

cc @jaeilepp @mainakjas

Eric89GXL, 2015-06-16:
From [here] (http://stackoverflow.com/questions/12439588/how-to-maximize-a-plt-show-window-using-python) it seems like the best solution is to add something like this to our keypress functions:
```
mng = plt.get_current_fig_manager()
mng.full_screen_toggle()
```
@jaeilepp can you give it a try in your next round of changes?

Eric89GXL, 2015-06-16:
My vote is for this to be tied to F11

dengemann, 2015-06-16:
Sounds good, also an option at call time would be great to let it start like that.

agramfort, 2015-06-16:
I'd rather remember the size of the window in the config rather than
default to full screen.

Eric89GXL, 2015-06-16:
@agramfort I don't think we want to default to full screen, either, it should default to `False` if we add a kwarg.

Eric89GXL, 2015-06-17:
Leaving this one open, but changing the title to reflect it's now an issue with OSX

Eric89GXL, 2017-03-21:
Closing as MPL bug (if it's still happening for OSX backend)

closed 2017-03-21 by Eric89GXL