mne-tools/mne-python #2213: GSOC TODO list

jaeilepp, 2015-06-18:
In a discussion with @dengemann and @mainakjas, we decided to make a list of things to do over this GSOC. Feel free to add items to the list. Preferably stuff about the vizualization, since that's my field for this GSOC.

teonlamont, 2015-06-18:
These are html visualizations/interactions with MNE-Report that could be improved upon:

MNE-Report toggle button https://github.com/mne-tools/mne-python/issues/1996
MNE-Report html issue https://github.com/mne-tools/mne-python/issues/2212

agramfort, 2015-06-18:
time frequency interactive viz e.g. select region of tf plane and get the
associated topography (see how fieldtrip does it).

maybe interactive selection of bad ICA components (via topo or time courses)

mainakjas, 2015-06-18:
@teonlamont I can take care of those two issues so that @jaeilepp can focus on core functionality.

mainakjas, 2015-06-18:
filing one idea we had discussed with @dengemann : topoplot with sliders (for different time points) in mne-report (like the bem plot). Good to visualize evolution of evoked response

mainakjas, 2015-06-18:
... also interactive picking of channels from a plotted layout. @teonlamont can give a demo maybe how he uses it.

teonlamont, 2015-06-18:
also, it would be nice if we could have a standalone viewer of Montage/DigMontage class. it's essentially already implemented in kit2fiff gui. it would just need the removal of the KIT input file. The reason it would be useful to have it is that if you are interested in your `dev_head_t`, you can then visualize the headshape info and the device hpi coil points. there are often cases where hpi coils fall off and the `dev_head_t` would be affected by it.

teonlamont, 2015-06-18:
and @mainakjas i'm looking forward to your pr's ;)

kingjr, 2015-06-18:
> filing one idea we had discussed with @dengemann : topoplot with sliders (for different time points) in mne-report (like the bem plot). Good to visualize evolution of evoked response

What I sometimes do is making animated gif. They're easy to email or put them in powerpoints slides. It'd be great to have this within the package, although I guess it's a bit vintage.

dengemann, 2015-06-18:
> also, it would be nice if we could have a standalone viewer of Montage/DigMontage class. it's essentially already implemented in kit2fiff gui. it would just need the removal of the KIT input file. [...]

maybe not directly related to Jaakko's GSOC, no?

dengemann, 2015-06-18:
I'll list the points in a more  structured way:

For existing functions:
- [ ] add scale + unit information on demand to browser plots
- [ ] add option to hide channel names for views with more many channels, e.g. > 100. Maybe auto-hide
- [ ] infobox with more precise control options, e.g. add 40 channels, add 22 epochs, etc.

Next/Maybe:
- [ ] scrollable topomps in matplotlib / report
- [ ] interactive TFR plotting
- [ ] interactive ICA selection

dengemann, 2015-06-18:
To keep the discussion compact, please edit my post above to add items.

teonlamont, 2015-06-19:
+1 on the scrollable plots/slider. it should be general so that you can take any series of images and turn them into a slider. currently dealing with ROC curves over time and it would be great to have a way to consolidating all the plots into one figure.

dengemann, 2015-06-22:
updated todo @jaeilepp

jaeilepp, 2015-06-26:
I can't seem to update the checkboxes on @dengemann comment.
- [ ] add scale + unit information on demand to browser plots
- [ ] add option to hide channel names for views with more many channels, e.g. > 100. Maybe auto-hide
- [ ] infobox with more precise control options, e.g. add 40 channels, add 22 epochs, etc.
- [ ] make sure all interactive options can be initialized from the function call
- [ ] option to turn off event start line
- [ ] evoked.plot_topo cf #2240 (added by mainak)

Next/Maybe:
- [ ] scrollable topomps in matplotlib / report
- [ ] interactive TFR plotting
- [ ] interactive ICA selection

jaeilepp, 2015-06-26:
> make sure all interactive options can be initialized from the function call

Do you mean that for example butterfly plot should be operated with a boolean mask (butterfly=True)?

dengemann, 2015-06-26:
Yes or view/mode='butterfly' this would also help avoid the trellis param

Eric89GXL, 2015-11-25:
GSoC is over now, so I'm closing. But it's nice to see a mostly cleared long checklist above @jaeilepp :)

closed 2015-11-25 by Eric89GXL