mne-tools/mne-python #2218: bad channels not present

mainakjas, 2015-06-18:
If a channel is not present and I mark it as bad, MNE-Python lets me get away with it. This is not really desired -- makes debugging hard for user.

```py
import mne
from mne import io
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
tmin, tmax = -0.2, 0.5

event_id = {'Auditory/Left': 1, 'Auditory/Right': 2}

raw = io.Raw(raw_fname)
events = mne.read_events(event_fname)

raw.info['bads'] += ['EEG 0723']  # some random bad channel that doesn't exist

# pick EEG and MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=True, stim=False, eog=True,
                       exclude='bads')
# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(eeg=80e-6, eog=150e-6),
                    preload=True)
```

teonlamont, 2015-06-18:
I agree, there should be a check of whether the channel name added to the info['bads'] is in the info['ch_names']. pr welcome ;)

Eric89GXL, 2015-06-18:
I think we'll probably need to make `Info` smarter so we can keep the list of bad channels a simple list.

agramfort, 2015-06-18:
@mainakjas what do you suggest? error, warning? an Info.check() method?

mainakjas, 2015-06-18:
Error I suppose? I don't see a good reason to add a channel as bad if it doesn't exist. Not sure how it would be implementation-wise. Can we check whenever info['bads'] is modified?

dengemann, 2015-06-18:
I don't see an elegant solution here ...

- subclassing list and adding logic in ```__setitem__``` is not what we
want I guess ...
- a method for setting bads will not be used by anyone who already is used
to set the bads as a list
- checking if info has changed is expansive and complex e.g. with a second
thread .... , does not scale well given the problem

couldn't we just remove channels automatically from bads that are not in
the instance, when ever we look up the field somewhere down-stream?

dengemann, 2015-06-18:
We could of course overwrite __getitem__ in the Info and add checking logic
whenever 'bads' is looked up.

2015-06-18 21:32 GMT+02:00 Denis-Alexander Engemann <
denis.engemann@gmail.com>:

Eric89GXL, 2015-06-18:
Unfortunately I don't think subclassing either will work. Right now you can set bads any of these ways:
```
info['bads'] = ['EEG001', 'EEG002']
```
or
```
info['bads'] += ['EEG001', 'EEG002']
```
or
```
bads = ['EEG001']
info['bads'] = bads
bads += ['EEG002']
```
Subclassing will fix only the first and second of these cases. We don't know what users have been doing, so we can't guarantee all fixes will work.

What we can do is make it so that any time we do an operation that involves `bads`, we do a quick `_check_info(info)` that does some consistency checks. For now it would just be that `bads` exist and maybe another simple check like:
```
def _check_info(info):
    if any(bad not in info['ch_names'] for bad in info['bads']):
        raise RuntimeError('bad channel marked does not exist in info...')
    if set(info['ch_names']) != set(ch['name'] for ch in info['chs']):
        ...
```
It makes it so the notification will come later, e.g. when doing something with the data or saving it, but that seems like the least bad solution to me.

dgwakeman, 2015-06-18:
Is it hard to make it impossible to do the third example? 

We could start deprecating it and implement the checking @dengemann described. It also follows the preferably one simple way of doing things: IMO 1 and 2 do different and useful things, but 3 is just another way of doing things

Eric89GXL, 2015-06-18:
You can can't make the third example impossible, no. You can change the behavior by making the last line that adds a bad channel not do anything, but that could break people's code -- and worse, it would do it silently, which is unsafe. Unfortunately our use of built-in mutable datatypes of dict and list allow for doing things multiple ways already, so we can't follow the one-way mantra without breaking people's code...

dengemann, 2015-06-18:
Not sure. Just make sure it can be used as before. I'm -1 about changing
the API. If we can elegantly implement checking / sanitizing we should do
it, if not this prolem is not severe enough to justfy the efforts.

agramfort, 2015-06-18:
I would go with the _check_info approach that is called when changing an
instance / making a copy. Sounds ok?

what else do we want to check besides bads?

dengemann, 2015-06-18:
> I would go with the _check_info approach that is called when changing an

+1

> what else do we want to check besides bads?

really not sure.

Eric89GXL, 2015-06-18:
There are probably lots of things we could check in principle, but I think YAGNI until we see otherwise. I think checking bads and that `info['ch_names']` matches the `info['chs']` might be enough to start.

When you say "changing an instance", what do you mean? We can't easily detect the various use cases that users might have for changing the `info['bads']` property as in my example above, hence why I suggested adding the check anywhere that *uses* the 'bads' property. Although if we make the checking function more general, we might eventually want to make the function more widespread, and also do it e.g. when making a copy.

agramfort, 2015-06-18:
I would add it the few epochs methods that seem relevant

mainakjas, 2015-06-19:
sounds like a plan. 

Should we also have methods to update `info`. And these methods have proper validation checks -- e.g., `update_bads` and  we update the examples accordingly to use this? Just throwing ideas ...

dengemann, 2015-06-19:
Sounds a bit of overkill to me, just using info['bads'] as a list is very
neat.
An info method is really verbose and I don't see why it's worth changing
API and usage recommandations for that.

dengemann, 2015-06-19:
Can anyone btw. tell me one example where this "feature" actually created a
bug?

2015-06-19 15:05 GMT+02:00 Denis-Alexander Engemann <
denis.engemann@gmail.com>:

mainakjas, 2015-06-19:
Imagine if you had a bad channel called `EEG 061` and you accidentally had a typo and put it as `EEG 0061`. You thought you got rid of the bad channel but actually you didn't. It silently went unnoticed and your data gets corrupted.

dengemann, 2015-06-19:
if it's really a bad channel it will hardly get unnoticed ;)
You will see it in the cov plots or elsewhere.
Anwyways, I'm fine with checking but against extra API.
Out of curiosity Mainak, is this a potential issue or did you have to deal
with it recently?

mainakjas, 2015-06-19:
Yeah, you will eventually notice it if you are careful with things. But I probably don't want anyone else to go through half an hour of debugging for the same reason. Yeah, I did have this issue myself -- so it's not just academic curiosity :) @teonlamont and @yousrabk will attest that I hit a few bugs / issues with mne-python just when I was in deadline mode ;)

dengemann, 2015-06-19:
> Yeah, you will eventually notice it if you are careful with things.

I would recomend a plot like this:

[image]

If you miss a bad channel the plot will be screwed up and in the grey image all structure will be lost.

> will attest that I hit a few bugs / issues with mne-python just when I was in deadline mode ;)

I believe you, I was just asking because I never ran into this one in ~3 years, at least not because of manual specification.

mainakjas, 2015-06-19:
> Yeah, you will eventually notice it if you are careful with things.
>
> I would recomend a plot like this:
>
> [image: screenshot 2015-06-19 15 35 40] [...]
Thanks. But this won't show flat channels or channels with oscillation
(e.g., we had a channel with 10 Hz oscillation in Helsinki). Or will it?

dengemann, 2015-06-19:
> Thanks. But this won't show flat channels or channels with oscillation
(e.g., we had a channel with 10 Hz oscillation in Helsinki). Or will it?

It should give you a grey line.
But would maybe need some tweaking / optimizitation, probably based on the color map. 
You could construct a second plot which shows you flat channels based on std or so.
Take home message is: include diagnostic plots in your data analysis routines that highlight undetected bads.

dengemann, 2015-06-19:
The more likely case is that you don't make a typo but you bad channels
detection routine is not perfect.
Checking the info won't help you there.

jaeilepp, 2015-07-10:
I noticed that the ``raw.info`` lets you get away also with duplicate entries of a bad channel. This actually happens in one of our examples. In ``plot_from_raw_to_epochs_to_evoked`` channel ``EEG 053`` is already there when it is added. I removed the manual adding in https://github.com/mne-tools/mne-python/pull/2251.

agramfort, 2015-07-11:
the idea of this is line is to show to people how to mark a bad channel.
It's basically documentation and should not be a problem for our
implementations. Please put it back.

dengemann, 2015-09-03:
What about this? Did we converge? Is it still hot? Will 0.11 it.

Eric89GXL, 2015-09-03:
I could try implementing the `_check_info` function quickly and see how badly it breaks our code :)

PR #2452 (2015-09-07): MRG: Check info consistency
Closes #2218.

Added some consistency checking of `info` regarding bads and found a few minor bugs along the way.
closed 2015-09-07 by agramfort