mne-tools/mne-python #2219: Q: Our package stats and statsmodels

Eric89GXL, 2015-06-18:
Has anyone tried using statsmodels for stats functions? There is discussion on the scipy dev list about deprecating the scipy anova functions, since they are limited and statsmodels already provides more functionality.

@dengemann have you thought about moving your n-way repeated measures anova over there, or just using their function if they have one?

It seems like integrating or facilitating statsmodels support could be useful going forward.

dengemann, 2015-06-18:
> @dengemann have you thought about moving your n-way repeated measures anova over there, or just using their function if they have one?

Hehe, I kind of triggered that debate, proposing to add out M-way ANOVA there. I predict poeple will suggest adding it to statsmodels, also because it has been repeatedly demanded by users and statsmodels does not have any repeated measure functions. That will requie some extra code to please their API, etc. Not really motivated to do that work. Would be more important to add support for between-subject factors to have mixed-effects models.

Eric89GXL, 2017-02-16:
FWIW I think statsmodels now has an ANOVA PR underway. Closing

closed 2017-02-16 by Eric89GXL