mne-tools/mne-python #2220: BUG: VIZ: evoked.plot_image(axes=ax)

kingjr, 2015-06-18:
I'm encoutering a bug when plotting an evoked image on a subplot if there are more than 3 subplots on the horizontal axis. I don't know how to fix this. Anyone with the same issue?

```
import numpy as np
import matplotlib.pyplot as plt
import mne
from mne import io
from mne.datasets import sample
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
raw = io.Raw(raw_fname, preload=True)
raw.filter(1, 20, method='iir')
events = mne.read_events(event_fname)
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=True, eog=True,
                       exclude='bads')
epochs = mne.Epochs(raw, events, dict(aud_l=1, vis_l=3), -.4, 1.2, proj=True,
                    picks=picks, baseline=None, preload=True,
                    reject=dict(grad=4000e-13, eog=150e-6)) 
evoked = epochs.average()
```

This works nicely

```
fig, ax = plt.subplots(2, 3)
for ii in range(3):
    evoked.plot_image(axes=ax[:, ii], show=False)
```

But this crashes painfully

```
fig, ax = plt.subplots(2, 4)
for ii in range(4):
    evoked.plot_image(axes=ax[:, ii], show=False)
```

Here is the error
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-20-5471963b487f> in <module>()
      1 fig, ax = plt.subplots(2, 4)
      2 for ii in range(4):
----> 3     evoked.plot_image(axes=ax[:, ii], show=False)

/home/jrking/anaconda/lib/python2.7/site-packages/mne/evoked.pyc in plot_image(self, picks, exclude, unit, show, clim, xlim, proj, units, scalings, titles, axes, cmap)
    435                                  show=show, clim=clim, proj=proj, xlim=xlim,
    436                                  units=units, scalings=scalings,
--> 437                                  titles=titles, axes=axes, cmap=cmap)
    438 
    439     def plot_topomap(self, times=None, ch_type=None, layout=None, vmin=None,

/home/jrking/anaconda/lib/python2.7/site-packages/mne/viz/evoked.py in plot_evoked_image(evoked, picks, exclude, unit, show, clim, xlim, proj, units, scalings, titles, axes, cmap)
    333                         hline=None, units=units, scalings=scalings,
    334                         titles=titles, axes=axes, plot_type="image",
--> 335                         cmap=cmap)
    336 
    337 

/home/jrking/anaconda/lib/python2.7/site-packages/mne/viz/evoked.py in _plot_evoked(evoked, picks, exclude, unit, show, ylim, proj, xlim, hline, units, scalings, titles, axes, plot_type, cmap)
    181                                extent=[times[0], times[-1], 0, D.shape[0]],
    182                                aspect='auto', cmap=cmap)
--> 183                 cbar = plt.colorbar(im, ax=ax)
    184                 cbar.ax.set_title(ch_unit)
    185                 ax.set_ylabel('channels (%s)' % 'index')

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/pyplot.pyc in colorbar(mappable, cax, ax, **kw)
   2152         ax = gca()
   2153 
-> 2154     ret = gcf().colorbar(mappable, cax = cax, ax=ax, **kw)
   2155     draw_if_interactive()
   2156     return ret

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/figure.pyc in colorbar(self, mappable, cax, ax, use_gridspec, **kw)
   1494         if cax is None:
   1495             if use_gridspec and isinstance(ax, SubplotBase):
-> 1496                 cax, kw = cbar.make_axes_gridspec(ax, **kw)
   1497             else:
   1498                 cax, kw = cbar.make_axes(ax, **kw)

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/colorbar.pyc in make_axes_gridspec(parent, **kw)
   1215 
   1216     parent.set_subplotspec(gs[0])
-> 1217     parent.update_params()
   1218     parent.set_position(parent.figbox)
   1219     parent.set_anchor(panchor)

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/axes/_subplots.pyc in update_params(self)
    113         self.figbox, self.rowNum, self.colNum, self.numRows, self.numCols = \
    114             self.get_subplotspec().get_position(self.figure,
--> 115                                                 return_all=True)
    116 
    117     def is_first_col(self):

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/gridspec.pyc in get_position(self, fig, return_all)
    423 
    424         figBottoms, figTops, figLefts, figRights = \
--> 425                     gridspec.get_grid_positions(fig)
    426 
    427 

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/gridspec.pyc in get_grid_positions(self, fig)
     84         nrows, ncols = self.get_geometry()
     85 
---> 86         subplot_params = self.get_subplot_params(fig)
     87         left = subplot_params.left
     88         right = subplot_params.right

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/gridspec.pyc in get_subplot_params(self, fig)
    370                            top=top,
    371                            wspace=wspace,
--> 372                            hspace=hspace)
    373 
    374         return sp

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/figure.pyc in __init__(self, left, bottom, right, top, wspace, hspace)
    184 
    185         self.validate = True
--> 186         self.update(left, bottom, right, top, wspace, hspace)
    187 
    188     def update(self, left=None, bottom=None, right=None, top=None,

/home/jrking/anaconda/lib/python2.7/site-packages/matplotlib/figure.pyc in update(self, left, bottom, right, top, wspace, hspace)
    219             if self.left >= self.right:
    220                 reset()
--> 221                 raise ValueError('left cannot be >= right')
    222 
    223             if self.bottom >= self.top:

ValueError: left cannot be >= right
```

yousrabk, 2015-06-19:
I think this is more a matplotlib problem when you have some overlapping like here the subtitles are too long. 
This is working for me:
```
fig, ax = plt.subplots(2, 4, figsize=(18, 7))
for ii in range(4):    
    evoked.plot_image(axes=ax[:, ii], show=False, titles=dict(grad='grad', mag='mag'))
```

kingjr, 2015-06-19:
That works for me! Thanks @yousrabk

closed 2015-06-19 by kingjr