mne-tools/mne-python #2223: difference between fwd['sol']['row_names'] and fwd['info']['ch_names']

cjayb, 2015-06-19:
Hiya, full disclosure first: the following is **not** reproducible on the sample dataset, which is why I've placed a minimal (250 MB) dataset on:

https://dl.dropboxusercontent.com/u/14937634/fwd_sol_vs_info_ch_ordering.tar.gz

The script `demo_fwd_sol_vs_info_ch_ordering.py` illustrates the situation on an MEG+EEG dataset (Triux 306 + 75ch EEG). I've included 1 sec of raw data (both "true" raw with IAS on and the tSSS'd version with IAS-effects removed), a trans-file, a standard pre-computed 3L-BEM and an ico-6 source space.

I've tracked the bizarre exception thrown by `sensitivity_map` to a call to `proj.py#302: fwd = pick_types_forward(fwd, meg=False, eeg=True`, which gets the channel names screwed up. Specifically, I found that `fwd['sol']['row_names']` and `fwd['info']['ch_names']` have a different ordering!? The code in `pick_types_forward` assumes the ordering to be the same, however, using the same `sel`-parameter to slice both `'sol'` and `'info'`.

The data is acquired as usual---I cannot think of a step which would have been performed differently. Is there any reason why the `'sol'` and `'info'` dicts should be allowed to have a different ordering? I'll gladly debug further, but could use some pointers on how to proceed. /Chris

Eric89GXL, 2015-06-19:
I think the forward solution code is structured so that the forward solution gets ordered as MEG then EEG channels. Is that how your data are organized originally? (I don't have time right now to look myself, but I can check later today or Monday.)

In any case, it does sound like there is a bug because if e.g. EEG data come before MEG data, the orders will be mismatched.

cjayb, 2015-06-19:
The `raw.info['ch_names']` is in the usual order
`['EOG001',
 'ECG002',
 'EOG003',
 'EEG001',
 'EEG002',
 'EEG003',
 'EEG004', ...]`, _i.e._, "BIO", followed by EEG, followed by MEG-chans. This is the order in `fwd['info']`. But in `fwd['sol']` the order seems to have been switched to MEG followed by EEG (BIO dropped, of course).

I'm sure you're right about the non-bug status, but if you do get the chance to take a look, I'd greatly appreciate it.

Eric89GXL, 2015-06-19:
I mean that there definitely is a bug... `fwd['info']` and `fwd['sol']` orders do not have to match -- we should use different selections for each, as you suggest. Do you want to try making a fix?

dgwakeman, 2015-06-19:
A good place to try things out would be to see what happens when you use the C code to deal with these files in each place (are the outputs ordered the same?)

Eric89GXL, 2015-06-19:
I suspect they will be since I tried to follow the C code closely with the forward code, but the C code probably does a better job with the picking after the fact :)

cjayb, 2015-06-19:
Sorry, I should learn to read _every_ word people write to me ;)

If it's safe to use separate selections, then at least for `pick_forward_types`, this  seems to work: 04bf94c104c7210352df5d0a04b39588164620ef 

I can't judge if it's "safe", though, any thoughts?

Eric89GXL, 2015-06-19:
This is about what I would expect. Does this work for your data? If so, please open a pull request.

We should also add some test, though. I think we can do it by loading a sample file, reordering the data and entries in `info`, and using that to create the forward. Are you up for writing that? If you don't have time or don't know how, I can do it.

cjayb, 2015-06-19:
It works to the extent that 1) the `sensititivity_map` is calculated and 2) the map "makes sense". I'd like to give this a go, I'll `@` you if/when need be (most likely in the test-phase). Expect a PR during next week. Have a good one!

Eric89GXL, 2015-06-19:
Great, thanks @cjayb.

In your PR on GitHub, please put `Closes #2223` so this issue gets auto-closed when me merge your PR.

PR #2229 (2015-07-06): allow different channel ordering in fwd['sol'] and fwd['info']
@Eric89GXL Here is the PR for the actual implementation. I've made a gist testing your suggested raw.info-reordering based on sample data:

https://gist.github.com/cjayb/60ce9a543a4665348ad4

The new ordering is based on the order I found in Triux raw files (EOG, EEG, MEG, STI), and reproduces the error. Having separate selections for sol and info dicts (@3ebdb09eba52ac7cd37089dc7d448f0b97c0bab8) indeed solves the situation, though I am a little uncertain about `fwd['_orig_sol']` and `fwd['_orig_sol_grad']`?

I tried to grasp the t [... cut]
closed 2015-07-06 by agramfort