mne-tools/mne-python #2224: ASCII format event file not compatible with mne_browse_raw

haribharadwaj, 2015-06-19:
Is this intentional? 
It does seem like we need two changes for compatibility (or update mne_browse_raw):

(1) Adding time column
(2) Adding dummy row with ```raw.first_samp``` (OR) adding ```raw.first_samp```  to all rows

agramfort, 2015-06-20:
can you clarify how the problem happens? what's your workflow?

haribharadwaj, 2015-06-20:
Take any events array and do ```mne.write_events('something.eve')```. The result cannot be viewed by doing "Load events from text file.." in ```mne_browse_raw```.

teonlamont, 2015-06-20:
I think it has something to do with the standard C `-eve.txt` file having four columns instead of our three if I remember correctly.

haribharadwaj, 2015-06-20:
It does..once I made the changes in OP, it shows up correctly.

Eric89GXL, 2015-06-20:
I thought you'd need the first_samp, too -- did it work without it?

haribharadwaj, 2015-06-20:
@Eric89GXL .. You don't technically need to have a dummy line with ```first_samp``` -- mne_browse_raw recognizes the the lack of a dummy line and shifts the sample numbers using the ```first_samp```  but that is incorrect because mne-python event arrays already have the requisite shifts.

Essentially you need whatever I have listed in OP.. 

Do you guys see anything breaking if I change mne.write_events to be compatible with mne_browse_raw ? If not, I can start a PR right away..

Eric89GXL, 2015-06-20:
You can try it first and see if it breaks any of the tests :)

I thought we had some reason for omitting the events, but looking back at the PR #111 I don't see one. So +1 for me

agramfort, 2015-06-21:
no objection. I never use events as texts file and I have never suggested
users to use them in the MNE training sessions.

Eric89GXL, 2015-11-25:
@haribharadwaj you still interested in tackling this?

haribharadwaj, 2015-11-25:
@Eric89GXL ..Sorry about dropping the ball on this.. I can do it but not for a couple of weeks. If anyone else can do it earlier, they should feel free

Eric89GXL, 2017-02-16:
Since this doesn't seem to be in high demand I'll close this. But @haribharadwaj if you would still like to see it feel free to reopen and hopefully someone can get to it

closed 2017-02-16 by Eric89GXL