mne-tools/mne-python #2225: BUG make_fwd_solution is broken

teonlamont, 2015-06-21:
I was trying to make a fwd solution and I got an error.

```python
/Applications/packages/mne-python/mne/io/pick.pyc in pick_types(info, meg, eeg, stim, eog, ecg, emg, ref_meg, misc, resp, chpi, exci, ias, syst, seeg, include, exclude, selection)
    191     from .meas_info import Info
    192     if not isinstance(info, Info):
--> 193         raise TypeError('info must be an instance of Info, not %s' % info)
    194     nchan = info['nchan']
    195     pick = np.zeros(nchan, dtype=np.bool)

TypeError: info must be an instance of Info, not {'dev_head_t':...```

it's because the internal dict that is being generated is not an instance of Info, just a regular dict. this is a problem when you do pick_types on it.

cc: @agramfort, @Eric89GXL

agramfort, 2015-06-21:
Can you send a PR?

Eric89GXL, 2015-06-21:
@teonlamont in fact I think you must be on my branch in order to have gotten this error -- I added that check in my PR. Please check the most recent version of my PR to see if it fixes your problem, and if it does, close your PR.

teonlamont, 2015-06-21:
yeah, life living on multiple branches. yeah, this fixes the error I was having. I will close this issue and my pr, thanks

PR #2211 (2015-06-27): MRG: Refactor Epochs classes
Not working yet. I hope to keep this more red than green. To-do list:

* [x] Move methods to `_BaseEpochs`
* [x] Make `EpochsArray` subclass `_BaseEpochs` instead of `Epochs`
* [x] Refactor how projection is done (move to `_BaseEpochs` constructor; #2200)
* [x] Refactor how decimation is done (add method, call it in constructor; #2178)
* [x] Refactor how rejection/dropping is done (add method; #1893)
* [x] Get EpochsArray working again
* [x] Fix RtEpochs
* [x] Fix EpochsKit, make it subclass `_BaseEpochs` instead of `EpochsArray` (which in turn had been subclassing [... cut]
closed 2015-06-21 by teonlamont