mne-tools/mne-python #2230: Getting Epochs Indices after selection, then rejection

teonlamont, 2015-06-22:
Is there a simple way to get the index of the epochs that you drop other than drop log? here's the situation.

I have an Epochs and behavioral data, same length.
If I do event selection based on the behavioral data, it's fine. I just get the indices and submit them to epochs.drop_bads. The two are still the same length. The drop log however is the original length.
The problem is if I decide to reject epochs using a reject parameter (mag=3e-12), I don't have a way of knowing which epochs were dropped with this reduced epochs because the drop log is the original size but the behavioral and the new reduced epochs are the same length but smaller than the drop log.

dengemann, 2015-06-22:
epochs.selection

teonlamont, 2015-06-22:
that just gives me an np.arange of the original Epochs

dengemann, 2015-06-22:
> that just gives me an np.arange of the original Epochs

not sure why this is. epochs.selection should be the index relative to the events you pass to the constructor. I often used it to sync epochs to external events, e.g. behavioral data.

agramfort, 2015-06-22:
epochs.selection should do it

teonlamont, 2015-06-22:
so what happens is this: `epochs` and `beh_data` start at 1000: 
```
beh_data = beh_data[::2]
epochs.drop_bads(np.arange(1000)[::2])
epochs.drop_bad_epochs(reject={'mag': 2e-12})  # some unknown list of epochs dropped
beh_data = beh_data[?]  # unclear what idx to use to make comparable
```

teonlamont, 2015-06-22:
Oh, I see, I should not index the `beh_data` until the very end. that should work

dengemann, 2015-06-22:
@teonlamont what if you first drop bads based on rejection? I'm not sure if there might be an issue with epochs.selection after calling epochs.drop_bads. But epochs.selection should really do what you want.

teonlamont, 2015-06-22:
yep, that was the solution. Epochs should be dropped first but rejecting based on the behavioral data. then you can do the behavioral indexing without a problem. thanks for being a soundboard :)

closed 2015-06-22 by teonlamont

dengemann, 2015-06-22:
@teonlamont I don't fully get what was the solution. Could you clarify it a bit more? Would be good for the records.

teonlamont, 2015-06-22:
oh ok. so say you have two datasets `epochs` and `beh_data`. They both have 100 events. There are two conditions `left` and `right`, 50 each, for simplicity, first 50 are `left`, and the last 50 are `right`. 

If you want to keep them align, first you should reject first based on the meg rejection criteria. The reasoning is as follows (using pseudo-code):
```python
right = epochs['right']
right.selection
    array([50, 51, 52, ..., 99])
len(right)
    50
beh_data = beh_data[right.selection]

goods = [25, 63, 47, 18]  # assume these were the only good ones based on behavior
beh_data = beh_data[goods]
epochs = epochs[goods]
```
this is the correct order for threshold rejection or epochs selection via event_id.

but if you were to rejection based on `beh_data` first:
```python
idx = np.ones(100)
idx[[4, 7, 53]] = 0
beh_data = beh_data[idx]
len(beh_data)
    97
epochs = epochs[idx]
len(epochs)
    97
epochs.selection
    array([0, 1, 2, 3, 5, 6, 8, 9, 10, ..., 52, 54, 55, ..., 98, 99])
epochs.drop_bad_epochs(reject=3e-12)  # assume that the first epochs dropped for simplicity
epochs.selection
    array([2, 3, 5, 6, 8, 9, 10, ..., 52, 54, 55, ..., 98, 99])

beh_data = beh_data[epochs.selection]  # this will not work because indexing is based on orig
```

The selection indices refer to the epochs index of the original epochs that you started it with so you would not be able to use these indices with beh_data because the indices won't be aligned at this point.

dengemann, 2015-06-22:
Thanks @teonlamont that is absolutely what I supsected. Seemingly there was a misnomer in your previous description. Thanks for the clarification, should help others.

dgwakeman, 2015-06-22:
This may seem silly, but isn't it better to just change the triggers on the raw data file prior to epoching. Using your easy example:
Left is code 1
Right is code 2
Simply add 4 to any "wrong" trial then epoch on 1 and 2
Then all 5s and 6s will be wrong, and if you want to do another epoch for example not caring about behavioral performance simply use ignore=4

dengemann, 2015-06-22:
@dgwakeman you can change your events before you construct epochs, sure.

dengemann, 2015-06-22:
I think these are complementary aspects., Ultimately you still might want to know which remaining epochs is where in the original sequence of events.