mne-tools/mne-python #2236: baseline correction too verbose

mainakjas, 2015-06-24:
I feel that baseline correction is too verbose:

```
Applying baseline correction ... (mode: mean)
Reading 1966985 ... 1969735  =    393.397 ...   393.947 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 1969485 ... 1972235  =    393.897 ...   394.447 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 1971985 ... 1974735  =    394.397 ...   394.947 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 1974485 ... 1977235  =    394.897 ...   395.447 secs...
[done]
```sh

I have thousands of trials and by the end of it, I lose any messages that were before this. If we can use this instead: https://github.com/mne-tools/mne-python/blob/b143c6df244dca2e6121048fae99cb6e1cfa84ab/mne/utils.py#L1148, that would be great.

agramfort, 2015-06-24:
can't you just set verbose=False for this function?

mainakjas, 2015-06-25:
of course I can :) but as @teonlamont would say, that would be a hack ;-)

teonlamont, 2015-06-25:
actually, in this case I wouldn't call it a hack since we have a proper implementation for dealing with this.

teonlamont, 2015-06-25:
maybe you want something like verbose='minimal' but that would be a bit overkill ;)

mainakjas, 2015-06-25:
no but seriously, I don't want to suppress the progress bar completely. I like to know where I am, just a little more streamlined :)

teonlamont, 2015-06-25:
I guess you're saying that pretty much no one needs to know that the hundreds of epochs you baseline was in fact baseline individually. I would agree with that.

teonlamont, 2015-06-25:
so you're suggesting making the output more concise and just overshare with your command line.

agramfort, 2015-06-25:
so don't plot the baseline correction for all epochs? how about the read
segments?

I personally work with 'warning' as log level

Eric89GXL, 2015-11-25:
@jasmainak any opinion on the "right" way to do it? I'm thinking just saying baseline is applied once (if it's easy enough to program), but leave in the `Reading` lines (since sometimes it can be slow) and get rid of the `[done]` since it's implied. WDYT?

kingjr, 2015-11-25:
What about a progress bar ?

jasmainak, 2015-11-25:
yeah, maybe I was thinking more along the lines of using the progressbar object in utils: https://github.com/mne-tools/mne-python/blob/62159a3dfbb1f0ca143958f2414d9c7497daccd7/mne/utils.py#L1192

Eric89GXL, 2015-11-25:
Progress bar doesn't get logged / is incompatible with `logger` / `logging` module. So for people who want to save their logs using `set_log_file` it won't work.

jasmainak, 2015-11-25:
Is it actually informative to have all those lines in a logging file? We can't have a one line summary for the logging file?

Eric89GXL, 2015-11-25:
I suppose we could do it. It's not backward compatible, but I strongly suspect nobody has been using the messages. Maybe we don't need the `Reading` messages at all, but instead whatever calls that should do the progress-bar'ing.

agramfort, 2015-11-25:
hum... why do you feel limited by set_log_file and passing
verbose=False whenever you need it?

jasmainak, 2015-11-25:
yes, that's possible but I feel this is not a good default. I don't see how printing "Applying baseline correction" for each data segment is informative. It simply obscures more important information on the console.

agramfort, 2015-11-25:
use verbose=False?
or introduce 2 levels of verbosity? making verbose an int

jasmainak, 2015-11-25:
yeah, I think introducing levels of verbosity might be a good idea

jona-sassenhagen, 2016-02-09:
>yes, that's possible but I feel this is not a good default. I don't see how printing "Applying baseline correction" for each data segment is informative. It simply obscures more important information on the console.

Agreed. As a beginner, this really annoyed me.

agramfort, 2016-02-11:
fait enough improvement suggestion welcome

Eric89GXL, 2016-02-22:
How about a simple fix of reporting the baseline correction type when instantiating the object only?

Eric89GXL, 2016-02-22:
Probably preferable to our own levels -- from the `logging` docs:

> Defining your own levels is possible, but should not be necessary, as the existing levels have been chosen on the basis of practical experience. However, if you are convinced that you need custom levels, great care should be exercised when doing this, and it is possibly a very bad idea to define custom levels if you are developing a library. That’s because if multiple library authors all define their own custom levels, there is a chance that the logging output from such multiple libraries used together will be difficult for the using developer to control and/or interpret, because a given numeric value might mean different things for different libraries.

So let's see if we can do without it.

PR #2926 (2016-02-23): MRG: Cleaner logging of baselines
Closes #2236.

Streamlines our logging messages about baselines a bit. Now it's done once per object, during instantiation / creation.
closed 2016-02-23 by agramfort