mne-tools/mne-python #2237: epochs.save doesn't split large epochs files

mainakjas, 2015-06-24:
When I try to save a large `epochs` I run into this error. I think @teonlamont also had a similar issue

cc @Eric89GXL 

```py
---------------------------------------------------------------------------
IOError                                   Traceback (most recent call last)
/tsi/doctorants/mainak/xyz.py in <module>()
    126                     picks=picks, baseline=(None, 0), reject=None,
    127                     detrend=1)
--> 128     epochs.save('test.fif')
    129     fix_stim_artifact(epochs, stim_channel='STI101')
    130     # epochs.drop_bad_epochs(reject=dict(eeg=200e-6))

/tsi/doctorants/mainak/mne-python/mne/epochs.pyc in save(self, fname)
   1366         data *= decal[np.newaxis, :, np.newaxis]
   1367 
-> 1368         write_float_matrix(fid, FIFF.FIFF_EPOCH, data)
   1369 
   1370         # undo modifications to data

/tsi/doctorants/mainak/mne-python/mne/io/write.pyc in write_float_matrix(fid, kind, mat)
    119     dims[-1] = mat.ndim
    120     fid.write(np.array(dims, dtype='>i4').tostring())
--> 121     check_fiff_length(fid)
    122 
    123 

/tsi/doctorants/mainak/mne-python/mne/io/write.pyc in check_fiff_length(fid, close)
    252         if close:
    253             fid.close()
--> 254         raise IOError('FIFF file exceeded 2GB limit, please split file or '
    255                       'save to a different format')
    256 

IOError: FIFF file exceeded 2GB limit, please split file or save to a different format
```

dengemann, 2015-06-24:
+1 for a fix!

Eric89GXL, 2015-06-24:
+1 as well, I won't have time to fix it in the next week, so please go for it if you want. You can look at what is done for Raw. I think we'll need to be able to split the `.data` attribute into multiple pieces when saving. In fact, while we're at it maybe we should change the `Epochs` file format so it saves each epoch independently. This would make not preload epochs much easier, too...

mainakjas, 2015-07-13:
Is anyone on this? I'm taking this if nobody is on this

Eric89GXL, 2015-07-13:
@mainakjas hop on gitter, I want to discuss an idea quickly

agramfort, 2015-07-17:
fixed by #2291

closed 2015-07-17 by agramfort