mne-tools/mne-python #2240: plot_topo should be a method of evoked?

mainakjas, 2015-06-25:
Silly question, but I was expecting to do:

```py
evoked.plot_topo()
```

instead of 

```py
from mne.viz import plot_topo
plot_topo(evoked)
```
anything against this? cc @jaeilepp interactive visualization is your department ;-)

teonlamont, 2015-06-25:
that would be useful

agramfort, 2015-06-25:
why not

dengemann, 2015-06-25:
The plot topo takes lists of evokeds too. I think that's why we kept it like that.

mainakjas, 2015-06-25:
I see, that makes sense but I guess it shouldn't be too hard to add a modified version of this function as a method to `evoked`. I use it all the time to look at my data.

agramfort, 2015-06-26:
it should not be too much lines of code indeed.

dengemann, 2015-06-26:
@mainak you could also subclass Evoked from list to make it work more
flawlessly ;)

Eric89GXL, 2015-10-19:
This one's done by #2241

closed 2015-10-19 by Eric89GXL