mne-tools/mne-python #2243: Help with make_forward_solution

ImmanuelSamuel, 2015-06-26:
mne.make_forward_solution(info, trans, src, bem, fname=None, meg=True, eeg=True, mindist=0.0, ignore_ref=False, overwrite=False, n_jobs=1, verbose=None, mri=None)

I am doing this on a EEG dataset. I have the epochs instance. I am suing the 128 Biosemi system. How to get trans, src and bem?
Thanks

Eric89GXL, 2015-06-27:
Have you looked through the MNE manual? The topics of what a trans is (coregistration matrix for head digitization and structural), src is (source space for neural currents), and what a BEM is (conductor model for forward calculation) are covered in there.

ImmanuelSamuel, 2015-07-07:
I have been looking into all of the above and it seems to be linux heavy in getting all the files necessary to do an analysis. Is it possible to have a standard source space, BEM and trans obtained from a standard T1? I don't have individual T1 so I was initially planning on running getting the files based on a standard T1 as used in MRIcron. Is it possible to have some standard defaults making it easier for users to run source connectivity?

dengemann, 2015-07-07:
If you don't have a coregistration I doubt source connectivity is a great
idea ...
You can coregister to fsaverage if this is what you want using the coreg
gui in python.
It would be better to have t1 if you want to make claims about source
connectivity, my 2 cents.

ImmanuelSamuel, 2015-07-08:
Hmmm ok. I followed LORETA tutorials and they seem to do the same with a standard T1. That is why I asked if there is a default template that I we can use. I mean if people accept that method I am hoping they should accept this also. Although that is just my opinion.

Eric89GXL, 2017-02-16:
@ImmanuelSamuel you should be able to use `mne coreg` to create a surrogate from an existing subject by scaling the MRI as necessary. I'll close this, but if it doesn't make sense, please reopen

closed 2017-02-16 by Eric89GXL